FormerLurker/ArcWelderLib #5: Lot of artifacts

furynick, 2020-07-14:
I tried post-processing on a circular part:
[image]

 but the result is very "noisy"
[image]

Original g-Code is attached
[CabochonEchelle.zip](https://github.com/FormerLurker/ArcWelderLib/files/4920399/CabochonEchelle.zip)

Command line is the simplest one :
ArcWelderConsole.exe CabochonEchelle.gcode CabochonEchelle-weld.gcode

Using Simplify3D 4.1.2 to slice, Repetier Server 0.93.1 to render.

FormerLurker, 2020-07-14:
My guess is that your viewer isn't working, though it could be some problem with c++ compilation on your system (I'm relatively new to cross-platform development).  Try viewing your gcode in simplify3d.  Here's what I see after conversion:

[image]

furynick, 2020-07-14:
You're right, g-Code looks fine with Simplify visualizer ... I'll try to print it.

furynick, 2020-07-14:
Simplify doesn't have gyroid nor archimedean (yet), waiting for v5 ...

Print successfull !

closed 2020-07-14 by furynick

FormerLurker, 2020-07-14:
You could try Slic3r (or Prusa Slicer) or Cura too.  Anyway, enjoy the app!