FormerLurker/ArcWelderLib #12: Implement firmware retraction detection

FormerLurker, 2020-11-06:
This is partially implemented currently.  Fully implement this.

TodWulff, 2020-11-06:
you looking at using a print start script (akin to how OctoLapse sets its retraction) to suck those out of the slicer, or to acquire the retraction settings in some other manner?

FormerLurker, 2020-11-07:
I don't think firmware retraction is usually controlled by the slicer, but if it is then a gcode will be added.  Octolapse recognizes the gcode and acts appropriate.  The slicer retraction settings would still apply in that case.

FormerLurker, 2020-11-07:
Excuse me.. arc welder.  Octolapse does too, lol.  Since arc welder doesn't need to retract, all that I need to know is that the filament is retracted.  Bonus points if I know how much because you never know what kind of stull a user will pull, like using firmware retract and the detracting manually ( i have seen this) so I don't know where the filament is without knowing the firmware settings.

TodWulff, 2020-11-07:
Yeah, I overlooked the 'FW' aspects of the issue's language... 

You're having to deal with a multitude of different firmwares which is a big undertaking.

Out of curiosity (for the marlin builds), I looked at configuration.h and configuration_adv.h.   Seems the extruder related settings in the FW is somewhat limited.  If there is value in getting them detailed, I can provide same for my printer's FW (Ender 3 v2).