FormerLurker/ArcWelderLib #17: Missing license

fieldOfView, 2020-11-14:
This project seems to be missing a license. Is the license the same AGPL as the ArcWelderPlugin?

FormerLurker, 2020-11-23:
Hi @fieldOfView!  This slipped through the cracks, sorry.  I will update the license as soon as I have a chance.  I intend to use AGPL2.

atsju, 2021-02-20:
Hello, this is more or less related but I suppose that you will get more and more issues here doe to the arc welder plugin beeing integrated to Cura. Consider adding an issue_template.md file to get good quality tickets.

FormerLurker, 2021-02-21:
@atsju, good idea.  I have done that for Octolapse, but pretty much everyone just ignores it.  However,  there is a really awesome feature coming soon that allows forms within the issue template, so I may wait for that.  I am also planning to add a bot at some point.

Just FYI, I wrote the core ArcWelder algorithm, the console app, and the OctoPrint plugin, but @fieldOfView wrote the cura plugin.  It's a difficult setup regarding who gets what issue, but we pretty much refer to each other's repositories as necessary, depending on the issue.

Oh, @fieldOfView, I really need to post the license, lol!  So many things..

atsju, 2021-02-21:
@FormerLurker it's really easy to add a licence. Just clic "add file" in Github interface and name it licence. A button to use one of the templates will appear.
[image]

The interface is really good with a simplified view of permissions/limitations/conditions. 

[image]

I was about to do a PR for you but you mentioned "AGPL2" and the template propose only AGPL3 or GPL2. So I suppose it's better to give you the tip than to propose the PR.

FormerLurker, 2021-02-21:
Yeah @atsju, I'll have to mirror the arc welder plugin license.  Thanks for the tips!