FormerLurker/ArcWelderLib #20: CPU load too hight during printing

adoeller, 2020-12-06:
When arc welding during printing, the print process comes to a stop or starts stutter.

I did no "top" during arc welding, beause ssh does not work on my raspberry anymore, but maybe a "nice" value for the welding process prio would mitigate the problem?

Cheers,
Alex

FormerLurker, 2020-12-06:
This has already been handled in the devel branch code, which I am working to release ASAP.  The new behavior is that as soon as a print starts, any processing gcode within ArcWelder is stopped and added back to the front of the processing queue.  When the print ends, the gcode in the queue will begin processing again.

FormerLurker, 2020-12-13:
Hey, since this is a duplication and in the wrong repo, I'm going to close this.  You can track the status of the next arc welder plugin release at http://github.com/formerlurker/arcwelderplugin.

Thanks!

closed 2020-12-13 by FormerLurker