FormerLurker/ArcWelderLib #21: ArcWelder and ArcStraithener from release binaries fail on Win10

DBa2016, 2020-12-11:
Both ArcWelder.exe and ArcStraightener.exe exit without producing any output (no error message either) on Windows 10. Is there any special instruction to be followed when launching them?

FormerLurker, 2020-12-11:
The mingw versions are broken.  Are you using the msvc build?  How are you running it?  Screen cap of oitput maybe?

DBa2016, 2020-12-11:
> 
> 
> The mingw versions are broken. Are you using the msvc build? How are you running it? Screen cap of oitput maybe?

Yes, using MSVC build, plain starting from command line. There is no output at all - after few seconds, the execution just returns to cmd.exe.

FormerLurker, 2020-12-11:
Can you paste in an example of how you are calling it please?

DBa2016, 2020-12-11:
[image]

FormerLurker, 2020-12-28:
@DBa2016, can you try just running it with no parameters and see if it complains?  Then, try running it with the --help flag.  

The exe does require the [Visual C++ Redistributable](https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads)  (I need to document this....), so maybe make sure that is installed, and if not download and install it from the link.  Let me know because I'm working on this all day.

FormerLurker, 2021-01-03:
@DBa2016, I just finished creating an exe for windows that has all the static libraries included in the file.  Try [downloading this build](https://github.com/FormerLurker/ArcWelderLib/suites/1759533024/artifacts/33499772) and see if this helps.  Let me know, thanks!

bvrielink, 2021-01-08:
@FormerLurker Where are we supposed to put those files? I found ArcWelder.exe here: C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\bin\win64. But there is no ArcStraightener.exe. Is it new? Also there is no lib directory. So do we create it under C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin or under C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\lib\win64?

bvrielink, 2021-01-08:
Tried to put the libs under C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\lib\win64 and C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\lib put don't see any ArcWelder thingy in Cura.

FormerLurker, 2021-01-08:
@bvrielink,

> Where are we supposed to put those files? I found ArcWelder.exe here: C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\bin\win64. But there is no ArcStraightener.exe. Is it new? 

This actually should be a question aimed at the Cura Arcwelder plugin, which I did not create and I do not maintain.  However, I believe I can answer this one.

The Cura plugin uses a binary produced by this repository:  ArcWelder.exe

That exec wraps the arc welder algorithm in an application that can be called from the command line.  The 'Cura ArcWelder Plugin' (again, not my software) calls ArcWelder.exe in order to do the conversions.

ArcStraightener.exe is another binary this repository produces that does the opposite (converts G2/G3 to G1), which you do not want.

The *.lib* files are pre-compiled libraries that others could use if they want to write software that integrates arc welder.  If you are not writing programs in C++, you don't need to worry about those.

> Tried to put the libs under C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\lib\win64 and C:\Users\XXX\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\lib put don't see any ArcWelder thingy in Cura.

So, again, I didn't write the cura plugin, and actually do not yet have it installed yet (been incredibly busy unfortunately).  I looked at the [official plugin page](https://github.com/fieldOfView/Cura-ArcWelderPlugin), and there are no usage instructions yet.  I can tell you how you can use the ArcWelder.exe file directly if you want.  Otherwise you may consider going to the Cura Arcwelder Plugin github page and submit a feature request for usage instructions?

bvrielink, 2021-01-08:
@FormerLurker Sorry. Didn't know someone else is maintaining it. Will look at the plugin page.

FormerLurker, 2021-01-08:
@bvrielink, NP, I would help if I knew more about it!  I am a bit ashamed that I don't, in fact.  I will try to get it running for myself ASAP.

fieldOfView, 2021-01-17:
@bvrielink look at this issue at the plugin repository:
https://github.com/fieldOfView/Cura-ArcWelderPlugin/issues/6
There's a development snapshot of the plugin there, which includes a dependency which may be missing from your installation.

mrx23dot, 2021-02-01:
Latest one runs on win10

FormerLurker, 2021-02-01:
@mrx23dot, thanks for getting back to me!  Going to close this out since it looks like the c++ Visual C++ Redistributable is not longer needed.  Joy!

closed 2021-02-01 by FormerLurker