FormerLurker/ArcWelderLib #27: ArcWelder fails with UnicodeEncodeError: 'ascii' codec can't encode  character

susisstrolch, 2020-12-18:
OS: Linux
Environment: Cura Version: 4.8 master (Mark Burdon), Cura-mb-master-x86_64-20201121.AppImage
Cura-ArcWelder plugin: https://github.com/5axes/Cura-ArcWelderPlugin
ArcWelderLib: d3376dd37104846f9125c06bfd42807e21859351

Welding a G-code containing Unicode characters (like '°C') fails (see also https://github.com/FormerLurker/ArcWelderPlugin/issues/84).
```
2020-12-18 08:28:30,458 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [130]: Writing to [/home/
 jorschiedt/Development/3D-Print/Things/Nefertiti/Nefertiti-aw-cooper.3mf]...
 2020-12-18 08:28:30,459 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.execute [99]: Already post processed
 2020-12-18 08:28:30,681 - ERROR - [MainThread] UM.Logger.logException [107]: Exception: Unable to write to file Nefertiti-aw-
 cooper: 'ascii' codec can't encode character '\xb0' in position 2814: ordinal not in range(128)
 2020-12-18 08:28:30,702 - ERROR - [MainThread] UM.Logger.logException [111]: Traceback (most recent call last):
 2020-12-18 08:28:30,716 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/home/markb/3dp/cura/cura-build.git/build/
 inst/lib/python3/dist-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 150, in _writeToDevice
 2020-12-18 08:28:30,726 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/tmp/.mount_Cura-m9VAAmF/usr/bin/plugins/
 LocalFileOutputDevice/LocalFileOutputDevice.py", line 137, in requestWrite
 2020-12-18 08:28:30,734 - ERROR - [MainThread] UM.Logger.logException [111]:     self.writeStarted.emit(self)
 2020-12-18 08:28:30,740 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/home/markb/3dp/cura/cura-build.git/build/
 inst/lib/python3/dist-packages/UM/Signal.py", line 219, in emit
 2020-12-18 08:28:30,747 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/home/markb/3dp/cura/cura-build.git/build/
 inst/lib/python3/dist-packages/UM/Signal.py", line 336, in __performEmit
 2020-12-18 08:28:30,759 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/home/markb/3dp/cura/cura-build.git/build/
 inst/lib/python3/dist-packages/UM/Signal.py", line 219, in emit
 2020-12-18 08:28:30,765 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/home/markb/3dp/cura/cura-build.git/build/
 inst/lib/python3/dist-packages/UM/Signal.py", line 332, in __performEmit
 2020-12-18 08:28:30,771 - ERROR - [MainThread] UM.Logger.logException [111]:   File "/home/strolch/.local/share/cura/master/
 plugins/Cura-ArcWelderPlugin/ArcWelderPlugin.py", line 186, in _filterGcode
 2020-12-18 08:28:30,778 - ERROR - [MainThread] UM.Logger.logException [111]:     temporary_file.write(joined_gcode)
 2020-12-18 08:28:30,786 - ERROR - [MainThread] UM.Logger.logException [111]: UnicodeEncodeError: 'ascii' codec can't encode
 character '\xb0' in position 2814: ordinal not in range(128)
 2020-12-18 08:28:31,570 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully

```

susisstrolch, 2020-12-18:
Addendum: this happens because I also use @5axes post processing script https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/GCodeDocumentation.py which creates a G-Code comment header like:
```
;------------------------------------Infill------------------------------------
; Infill Density                                         : 10 %
; Infill Pattern                                         : grid
; Infill Overlap Percentage                              : 10 %
; Gradual Infill Steps                                   : 0
;-----------------------------------Material-----------------------------------
; Flow                                                   : 90 %
; Initial Layer Flow                                     : 90 %
; Support Flow                                           : 90 %
; Printing Temperature                                   : 220 °C
; Printing Temperature Initial Layer                     : 220 °C
; Build Plate Temperature                                : 50 °C
; Build Plate Temperature Initial Layer                  : 60 °C
;------------------------------------Speed-------------------------------------
; Print Speed                                            : 60 mm/s
; Infill Speed                                           : 120 mm/s
```
So, as a temporary personal workaround I'll add a search & replace postprocessor which simply filters out the '°' character.

5axes, 2020-12-18:
In this case could be more simple to modify the script and remove the unit export
`new_line = new_line + GelValStr + " " + str(GetUnit)`  -> `new_line = new_line + GelValStr`

susisstrolch, 2020-12-18:
@5axes are you talking about the GCodeDocumentation script?

FormerLurker, 2020-12-18:
I just fixed a unicode issue not too long ago (oh, I see you linked to that).  Will take a look.

susisstrolch, 2020-12-18:
Hmm, in pyhelper I saw you use defines to distinguish between python version.
So it may happen that the python on my machine, which is used for compilation, has a different version than the py comming with the Cura appimage which in fact may lead to different results.

FormerLurker, 2020-12-18:
@susisstrolch, thanks for taking a look at the c code.  I am always happy to have another set of eyes.  However, that gets compiled within the setup.py that runs in OctoPrint's venv if you are using Octoprint (you aren't, are you?).  The pyhelper code doesn't get included in the console app's exe since it isn't a dependency.  I see you posted the version of arcwelderlib you are using.  Did you get that by running arcwelder.exe --help?

susisstrolch, 2020-12-18:
Yep, that was from linux console, running a self compiled version, via "ArcWelder --help".

FormerLurker, 2020-12-18:
@susisstrolch, 

What compiler did you use?  That would be helpful for testing.  I'm in the middle of setting up a remote debugger now.  What a pain, lol!

susisstrolch, 2020-12-19:
Ok, here are the steps I did to compile:
```
cd ArcWelderLib
mkdir build
cd build
cmake ..
```
And here's the cmake output:
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/strolch/Desktop/mntDevelop/ArcWelderLib/build
```
And here's the make output:
```
/usr/bin/cmake -S/home/strolch/Desktop/mntDevelop/ArcWelderLib -B/home/strolch/Desktop/mntDevelop/ArcWelderLib/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/CMakeFiles /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/mnt/develop/ArcWelderLib/build'
make -f GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/build.make GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/depend
make[2]: Entering directory '/mnt/develop/ArcWelderLib/build'
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/strolch/Desktop/mntDevelop/ArcWelderLib /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib /home/strolch/Desktop/mntDevelop/ArcWelderLib/build /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/DependInfo.cmake --color=
Scanning dependencies of target GcodeProcessorLib
make[2]: Leaving directory '/mnt/develop/ArcWelderLib/build'
make -f GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/build.make GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/build
make[2]: Entering directory '/mnt/develop/ArcWelderLib/build'
[  3%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/array_list.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/array_list.cpp.o -c /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib/array_list.cpp
[  7%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/circular_buffer.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/circular_buffer.cpp.o -c /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib/circular_buffer.cpp
[ 11%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/extruder.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/extruder.cpp.o -c /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib/extruder.cpp
[ 14%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/gcode_comment_processor.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/gcode_comment_processor.cpp.o -c /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib/gcode_comment_processor.cpp
[ 18%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/gcode_parser.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/gcode_parser.cpp.o -c /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib/gcode_parser.cpp
[ 22%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/gcode_position.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/gcode_position.cpp.o -c /home/strolch/Desktop/mntDevelop/ArcWelderLib/GcodeProcessorLib/gcode_position.cpp
[ 25%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/logger.cpp.o
cd /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib && /usr/bin/c++  -DHAS_GENERATED_VERSION -I/home/strolch/Desktop/mntDevelop/ArcWelderLib/build/GcodeProcessorLib/generated  -fPIC   -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/logger.cpp.o -c /home/strolc [... cut ...]
/usr/bin/cc -fPIC   -shared -Wl,-soname,PyArcWelder.so -o PyArcWelder.so CMakeFiles/PyArcWelder.dir/py_arc_welder.cpp.o CMakeFiles/PyArcWelder.dir/py_arc_welder_extension.cpp.o CMakeFiles/PyArcWelder.dir/py_logger.cpp.o CMakeFiles/PyArcWelder.dir/python_helpers.cpp.o  ../ArcWelder/libArcWelder.a ../GcodeProcessorLib/libGcodeProcessorLib.a /usr/lib/x86_64-linux-gnu/libpython3.8.so -lstdc++ -lm 
make[2]: Leaving directory '/mnt/develop/ArcWelderLib/build'
[100%] Built target PyArcWelder
make[1]: Leaving directory '/mnt/develop/ArcWelderLib/build'
/usr/bin/cmake -E cmake_progress_start /home/strolch/Desktop/mntDevelop/ArcWelderLib/build/CMakeFiles 0
```

FormerLurker, 2020-12-28:
@susisstrolch, what version of the code did you compile?  I ran that header through the ArcWelder.exe in windows and here is the result:

```
M83
; é
;------------------------------------Infill------------------------------------
; Infill Density                                         : 10 %
; Infill Pattern                                         : grid
; Infill Overlap Percentage                              : 10 %
; Gradual Infill Steps                                   : 0
;-----------------------------------Material-----------------------------------
; Flow                                                   : 90 %
; Initial Layer Flow                                     : 90 %
; Support Flow                                           : 90 %
; Printing Temperature                                   : 220 °C
; Printing Temperature Initial Layer                     : 220 °C
; Build Plate Temperature                                : 50 °C
; Build Plate Temperature Initial Layer                  : 60 °C
;------------------------------------Speed-------------------------------------
; Print Speed                                            : 60 mm/s
; Infill Speed                                           : 120 mm/s
```

That's pretty much what I was expecting.  My guess is this is an issue with the cura plugin, probably a py2/py3 compatibility issue perhaps?  Try downloading the latest pre compiled binary artifacts and see if they work for you and let me know.

fieldOfView, 2021-01-08:
https://github.com/5axes/Cura-ArcWelderPlugin is a fork of https://github.com/fieldOfView/Cura-ArcWelderPlugin. The latter has a this issue fixed here:
https://github.com/fieldOfView/Cura-ArcWelderPlugin/commit/2edee70f07aa493737bd5dff2ab5567ec167fd33

FormerLurker, 2021-01-08:
@fieldOfView, OMG, I totally missed that in the issue description.  Thank you so much for that great catch :)  I spent a lot of time trying to figure out what was going on here, lol!

@susisstrolch, I'm going to go ahead and close this since it is not related to the arc welder algorithm or binaries.  Thanks for posting!

closed 2021-01-08 by FormerLurker