FormerLurker/ArcWelderLib #28: [Feature Request] Embedded GCode Parameter/Cmd to cache and save AW Console Output to generated file.

TodWulff, 2020-12-23:
Good day.  The thought is a request to enable console output to be cached and then written into the output gcode generated by AW.  This behavior could triggered by an embedded gcode AW Script Comment - think of how you're catching slicer config by way of comments on Octolapse.  The reason for this is that when AW is called from the Cura Plugin, the console output is seemingly lost to the ethers once the tool completes running.  It would certainly be nice to be able to see the CLI parameters passed to it by the slicer plugin, and the dialog/debug content normally displayed in the console, when one reviews the final gcode generated by AWL.

FormerLurker, 2020-12-28:
There can be way more console output than gcode, so I don't think I want it to go there.  You could ask the cura plugin creator to allow output redirection, or you could call the exe yourself and redirect the output.  In windows the command would be:

```
ArcWelder.exe {parameters go here} >> arc_welder_console_output.txt
```

That will save the output to a file.

FormerLurker, 2021-01-03:
I'm going to close this out @TodWulff .  Sorry that I won't be implementing your feature request, but I want to thank you for proposing it!  Please feel free to request other features in the future!

closed 2021-01-03 by FormerLurker