FormerLurker/ArcWelderLib #31: Build artefacts for OSX crash with a segfault

fieldOfView, 2021-01-18:
The build artefacts for OSX are not working. This seems to be caused by the buildprocess, because if I create a build on my local system the executable works fine.

Here's the output from `lldb` trying to run the ArcWelder command on a simple gcode file:
```
Aldos-Mac:Desktop aldo$ lldb macOS/bin/ArcWelder test.gcode 
(lldb) target create "macOS/bin/ArcWelder"
Current executable set to 'macOS/bin/ArcWelder' (x86_64).
(lldb) settings set -- target.run-args  "test.gcode"
(lldb) run
Process 639 launched: '/Users/aldo/Desktop/macOS/bin/ArcWelder' (x86_64)
2021-01-18 11:17:03.002 - arc_welder.gcode_conversion - INFO - Source and target path are the same.  The source file will be overwritten.  Temporary file path: 719a288e-3d85-c237-7184-f1d585b68792.tmp
2021-01-18 11:17:03.002 - arc_welder.gcode_conversion - INFO - Processing Gcode
	Source File Path             : test.gcode
	Target File Path (overwrite) : 719a288e-3d85-c237-7184-f1d585b68792.tmp
	Temporary File Path          : 719a288e-3d85-c237-7184-f1d585b68792.tmp
	Resolution                   : 0.0500mm (+-0.02500mm)
	Path Tolerance               : 5.000%
	Maximum Arc Radius           : 1000000mm
	Min Arc Segments             : 0
	MM Per Arc Segment           : 1000000.000
	Allow 3D Arcs                : False
	Allow Dynamic Precision      : False
	Default XYZ Precision        : 3
	Default E Precision          : 5
	G90/G91 Influences Extruder  : False
	Log Level                    : INFO
	Hide Progress Updates        : False
Process 639 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x000000010001c054 ArcWelder`___lldb_unnamed_symbol211$$ArcWelder + 2132
ArcWelder`___lldb_unnamed_symbol211$$ArcWelder:
->  0x10001c054 <+2132>: movq   %rdi, -0x68(%rax)
    0x10001c058 <+2136>: movq   (%rcx,%rbx,8), %rdi
    0x10001c05c <+2140>: movq   %rdi, -0x60(%rax)
    0x10001c060 <+2144>: movq   0x8(%rdx,%rbx,8), %rdi
Target 0: (ArcWelder) stopped.
(lldb) 
```
Unfortunately, the build artifacts are not debug builds, so it is hard to see what exactly is going wrong.

FormerLurker, 2021-01-18:
@fieldOfView, I could generate a debug build I think.  Better still would be to download the latest devel code and run this on a debugger.  It should build on Visual Studio Code as long as you install the cmake extension and set the build to debug.  I really need to get myself a mac :(

I just configured a raspberry pi 4 to run visual studio code, and have been doing some debugging and performance testing there.  It was quite illuminating.

fieldOfView, 2021-01-19:
To be clear: when I clone the repo (master branch) and do a build on OS X, the resulting binary works just fine. It is just the automated builds that fail with a segfault and produce the above output when run in the lldb debugger.

>  I really need to get myself a mac :(

I tested this on a VM running on a Windows host, because I don't have a recent mac either. I did confirm that the automated binaries segfault and the binaries I build work on other users' systems as well.

FormerLurker, 2021-01-19:
> To be clear: when I clone the repo (master branch) and do a build on OS X, the resulting binary works just fine. It is just the automated builds that fail with a segfault and produce the above output when run in the lldb debugger.

That is excellent.  Can you capture the output of your compiler?  I'd like to see what compiler you are using, as well as the flags.  I tried to create an action to build with gcc instead of clang (I know nothing about clang), but for whatever reason it still build with clang (possibly due to caching).  Going to keep trying.

> I tested this on a VM running on a Windows host, because I don't have a recent mac either.

Yes, that's exactly what I need to do.  I have been having some problems getting that to work (apparently there is a bit more that needs to be done to get them to run on AMD based processors), but think I'll eventually figure it out.

Anyway, I'm willing to spend whatever time is necessary to get the mac builds to work without issue.

fieldOfView, 2021-01-21:
Here's a terminal dump of me checking out and building the ArcWelderLib repository (with some newlines added in for readability).

```
Aldos-Mac:~ aldo$ git clone https://github.com/FormerLurker/ArcWelderLib
Cloning into 'ArcWelderLib'...
remote: Enumerating objects: 393, done.
remote: Counting objects: 100% (393/393), done.
remote: Compressing objects: 100% (221/221), done.
remote: Total 971 (delta 293), reused 249 (delta 165), pack-reused 578
Receiving objects: 100% (971/971), 999.85 KiB | 1.73 MiB/s, done.
Resolving deltas: 100% (702/702), done.

Aldos-Mac:~ aldo$ cd ArcWelderLib/
Aldos-Mac:ArcWelderLib aldo$ mkdir build
Aldos-Mac:ArcWelderLib aldo$ cd build/

Aldos-Mac:build aldo$ /Applications/CMake.app/Contents/bin/cmake ..

-- The C compiler identification is AppleClang 9.0.0.9000038
-- The CXX compiler identification is AppleClang 9.0.0.9000038
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /Library/Frameworks/Python.framework/Versions/3.5/lib/libpython3.5m.dylib (found version "3.5.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/aldo/ArcWelderLib/build

Aldos-Mac:build aldo$ make

/Applications/CMake.app/Contents/bin/cmake -S/Users/aldo/ArcWelderLib -B/Users/aldo/ArcWelderLib/build --check-build-system CMakeFiles/Makefile.cmake 0
/Applications/CMake.app/Contents/bin/cmake -E cmake_progress_start /Users/aldo/ArcWelderLib/build/CMakeFiles /Users/aldo/ArcWelderLib/build//CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/build.make GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/depend
cd /Users/aldo/ArcWelderLib/build && /Applications/CMake.app/Contents/bin/cmake -E cmake_depends "Unix Makefiles" /Users/aldo/ArcWelderLib /Users/aldo/ArcWelderLib/GcodeProcessorLib /Users/aldo/ArcWelderLib/build /Users/aldo/ArcWelderLib/build/GcodeProcessorLib /Users/aldo/ArcWelderLib/build/GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/DependInfo.cmake --color=
Scanning dependencies of target GcodeProcessorLib
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/build.make GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/build
[  3%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/array_list.cpp.o
cd /Users/aldo/ArcWelderLib/build/GcodeProcessorLib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DHAS_GENERATED_VERSION -I/Users/aldo/ArcWelderLib/build/GcodeProcessorLib/generated -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.12 -fPIC -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/array_list.cpp.o -c /Users/aldo/ArcWelderLib/GcodeProcessorLib/array_list.cpp
[  7%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/circular_buffer.cpp.o
cd /Users/aldo/ArcWelderLib/build/GcodeProcessorLib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DHAS_GENERATED_VERSION -I/Users/aldo/ArcWelderLib/build/GcodeProcessorLib/generated -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.12 -fPIC -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/circular_buffer.cpp.o -c /Users/aldo/ArcWelderLib/GcodeProcessorLib/circular_buffer.cpp
[ 11%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/extruder.cpp.o
cd /Users/aldo/ArcWelderLib/build/GcodeProcessorLib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DHAS_GENERATED_VERSION -I/Users/aldo/ArcWelderLib/build/GcodeProcessorLib/generated -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.12 -fPIC -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/extruder.cpp.o -c /Users/aldo/ArcWelderLib/GcodeProcessorLib/extruder.cpp
[ 14%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/gcode_comment_processor.cpp.o
cd /Users/aldo/ArcWelderLib/build/GcodeProcessorLib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DHAS_GENERATED_VERSION -I/Users/aldo/ArcWelderLib/build/GcodeProcessorLib/generated -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.12 -fPIC -std=gnu++11 -o CMakeFiles/GcodeProcessorLib.dir/gcode_comment_processor.cpp.o -c /Users/aldo/ArcWelderLib/GcodeProcessorLib/gcode_comment_processor.cpp
[ 18%] Building CXX object GcodeProcessorLib/CMakeFiles/GcodeProcessorLib.dir/gcode_parser.cpp.o
[... cut ...] elder.dir/python_helpers.cpp.o  ../ArcWelder/libArcWelder.a ../GcodeProcessorLib/libGcodeProcessorLib.a /Library/Frameworks/Python.framework/Versions/3.5/lib/libpython3.5m.dylib ../ArcWelder/libArcWelder.a ../GcodeProcessorLib/libGcodeProcessorLib.a /Library/Frameworks/Python.framework/Versions/3.5/lib/libpython3.5m.dylib -lc++ 
[100%] Built target PyArcWelder
/Applications/CMake.app/Contents/bin/cmake -E cmake_progress_start /Users/aldo/ArcWelderLib/build/CMakeFiles 0
Aldos-Mac:build aldo$ 
```

FormerLurker, 2021-01-23:
Thank you VERY much @fieldOfView!  This was helpful.  I noticed some static linking that wasn't happening with clang, and did get g++ working (even though I previously though that failed).  Can you [try the artifacts here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/494477196) and see if you get the segfault please?  Fingers crossed...

FormerLurker, 2021-01-23:
Actually, I may have spoken too soon...  The binaries are identical in size, which likely means nothing has changed.  Also, the linking I thought I didn't see in the clang build was due to a case sensitive search, so it was already doing that.  It doesn't really look like specifying gcc did anything.  Going to have to go through the log line by line and see if I can spot any differences.

fieldOfView, 2021-01-23:
No dice, unfortunately.

```
Aldos-Mac:Desktop aldo$ ./ArcWelder --version

./ArcWelder  version: 1.0.0-13-gf9cef7e

Aldos-Mac:Desktop aldo$ ./ArcWelder test.gcode 
2021-01-23 17:19:51.004 - arc_welder.gcode_conversion - INFO - Source and target path are the same.  The source file will be overwritten.  Temporary file path: 719a288e-3d85-c237-7184-f1d585b68792.tmp
2021-01-23 17:19:51.004 - arc_welder.gcode_conversion - INFO - Processing Gcode
	Source File Path             : test.gcode
	Target File Path (overwrite) : 719a288e-3d85-c237-7184-f1d585b68792.tmp
	Temporary File Path          : 719a288e-3d85-c237-7184-f1d585b68792.tmp
	Resolution                   : 0.0500mm (+-0.02500mm)
	Path Tolerance               : 5.000%
	Maximum Arc Radius           : 1000000mm
	Min Arc Segments             : 0
	MM Per Arc Segment           : 1000000.000
	Allow 3D Arcs                : False
	Allow Dynamic Precision      : False
	Default XYZ Precision        : 3
	Default E Precision          : 5
	G90/G91 Influences Extruder  : False
	Log Level                    : INFO
	Hide Progress Updates        : False
Segmentation fault: 11
Aldos-Mac:Desktop aldo$ 
```

fieldOfView, 2021-02-07:
Here's a Mac OS build of V1.1.1
[ArcWelder.zip](https://github.com/FormerLurker/ArcWelderLib/files/5940130/ArcWelder.zip)

FormerLurker, 2021-02-07:
@fieldofview FTW!!!