FormerLurker/ArcWelderLib #32: Feedback for V1.1.0

FormerLurker, 2021-01-24:
Please let me know if you've used the new build of ArcWelder or any of the associated libraries.  I would love to know what you like, what you don't, what you love, and what you hate.  If you have discovered any bugs, please mention those here.  I may ask you to create a separate issue for any newly discovered bugs.

Thank you all so much!  Without you all I couldn't have done this, and I'm feeling really good about the core algorithm now.

fieldOfView, 2021-01-25:
Here's a new [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ArcWelderPlugin/zzHJfdU_udY4lWeMO9r9Ha-zSXKtZbLMRyRejKldEXtt/ArcWelderPlugin-v7.0.0-2021-01-25T11%3A08%3A03Z.curapackage?Expires=1926932894&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=N8jg1rq3JW0q%2BeRHxmIRNaq5Lb87wH0yEoUrNUB6jLMAkVWlnKGGSHsnnqWhVMa7VYikI6vDzdxju3JGl0QS4CUI8h3HCRoWdlrmaYBb6oU9rm2E8pknuTxHWEmNdTjEyAJ3FhKANYi2%2BoYE8NsoxAwA%2FLmXEKdr%2FGE8smqWV4qgWhJNv1eITT%2F8pKChkZR3MmFIF%2FzjHHjSLWks4UjLeX03IUYTrozi3tpv2E80ebfXDzEtQMvTALsVjoDSOByy7xWNId3zAC2iZAdxplzyW54z8pW3xOYLae3lM0cHLhcfbOJMBGLctntAehlXes0DmhLvEjptwbPCRvBylaHCGg%3D%3D) to try, including the latest version of the ArcWelder executables. It also includes a build for macOS: https://github.com/fieldOfView/Cura-ArcWelderPlugin/blob/bac9e81/bin/osx/ArcWelder

Nitpick: it is raspbian, not rasbian. But that name is officially no longer used; it is Raspberry Pi OS now.

FormerLurker, 2021-01-25:
Thanks for the spelling catch, and the new files!  I will take a look and upload after work.

FormerLurker, 2021-01-25:
@fieldOfView, I went ahead and attached your mac build (thank you so much for that!) and corrected the raspbian release (Raspberry_Pi_OS).  Looking much better now.  I'm looking forward to trying out the development snapshot.  Will provide feedback on that.

egar0, 2021-01-25:
Just downloaded v1.1.0 for my MacBook (MacOS 11.1) and it ran without a hitch.  I have not yet printed the resulting gcode, but looking over the gcode, I would expect it to work.  v1.0.0 did not run on my Mac and had the segfault error that is in issue #31.

FormerLurker, 2021-02-08:
Thanks everyone!  New binaries are up, including the MacOs build by @fieldOfView (thank you!), and a 32 bit windows version (might need the C++ runtime, not sure yet, don't have 32 bit os to test on yet).  Find the new feedback thread [here](https://github.com/FormerLurker/ArcWelderLib/issues/43)

closed 2021-02-08 by FormerLurker