FormerLurker/ArcWelderLib #34: Malformed model after welding

gkatev, 2021-02-04:
Hi,
I've encountered an issue, with the master branch of ArcWelderLib.
I did not see the problem with the Octroprint plugin (up to date with stable version I think).

Before welding (spacer.gcode):
[image]

After welding (spacer.aw.gcode):
[image]

Relevant files are here: [spacer.zip](https://github.com/FormerLurker/ArcWelderLib/files/5924765/spacer.zip)

```
$ build/ArcWelderConsole/ArcWelder ~/Desktop/spacer.gcode ~/Desktop/spacer.aw.gcode
2021-02-04 11:54:42.001 - arc_welder.gcode_conversion - INFO - Processing Gcode
	Source File Path             : /home/gkatev/Desktop/spacer.gcode
	Target File File             : /home/gkatev/Desktop/spacer.aw.gcode
	Resolution                   : 0.0500mm (+-0.02500mm)
	Path Tolerance               : 5.000%
	Maximum Arc Radius           : 1000000mm
	Min Arc Segments             : 0
	MM Per Arc Segment           : 0.000
	Allow 3D Arcs                : False
	Allow Dynamic Precision      : False
	Default XYZ Precision        : 3
	Default E Precision          : 5
	G90/G91 Influences Extruder  : False
	Log Level                    : INFO
	Hide Progress Updates        : False
2021-02-04 11:54:42.003 - arc_welder.gcode_conversion - INFO - arc_welder::process - Parameters received: source_file_path: '/home/gkatev/Desktop/spacer.gcode', target_file_path:'/home/gkatev/Desktop/spacer.aw.gcode', resolution_mm:0.05mm (+-0.03mm), path_tolerance_percent: 0.05, max_radius_mm:1000000.00, min_arc_segments:0, mm_per_arc_segment:0, g90_91_influences_extruder: False, allow_3d_arcs: False, allow_dynamic_precision: False, default_xyz_precision: 3, default_e_precision: 5
Progress:  percent_complete:0.00, seconds_elapsed:0.00, seconds_remaining:inf, gcodes_processed: 0, current_file_line: 0, points_compressed: 0, arcs_created: 0, num_firmware_compensations: 0, compression_ratio: 0.00, size_reduction: 0.00% 
Progress:  percent_complete:100.00, seconds_elapsed:0.04, seconds_remaining:0.00, gcodes_processed: 14292, current_file_line: 17723, points_compressed: 6513, arcs_created: 301, num_firmware_compensations: 0, compression_ratio: 1.43, size_reduction: 30.14% 
2021-02-04 11:54:42.044 - arc_welder.gcode_conversion - INFO - 
   Min          Max     Source  Target  Change
----------------------------------------------
  0.000mm to   0.002mm       0       0    0.0%
  0.002mm to   0.005mm       0       0    0.0%
  0.005mm to   0.010mm       0       0    0.0%
  0.010mm to   0.050mm       0       0    0.0%
  0.050mm to   0.100mm       0       0    0.0%
  0.100mm to   0.500mm    3321    1317  -60.3%
  0.500mm to   1.000mm    5430    2678  -50.7%
  1.000mm to   5.000mm    4539    4613    1.6%
  5.000mm to  10.000mm       4      77 1825.0%
 10.000mm to  20.000mm       0     147     INF
 20.000mm to  50.000mm       0       2     INF
 50.000mm to 100.000mm       0       0    0.0%
          >= 100.000mm       0       0    0.0%
----------------------------------------------
Total distance source:.............10493.418mm
Total distance target:.............11209.758mm
   Total count source:...................13294
   Total count target:....................8834
 Total percent change:..................-33.5%
2021-02-04 11:54:42.044 - arc_welder.gcode_conversion - INFO - Arc Welder process completed successfully.
```

FormerLurker, 2021-02-04:
Can you run the console version with the --help command and paste in the version?  I just want to double check to make sure I'm debugging the exact right one.  Thanks!

gkatev, 2021-02-04:
@FormerLurker
```
Version: 1.1.0, Branch: master, BuildDate:
   2021-02-04T09:54:13Z
```

FormerLurker, 2021-02-04:
OK, we are on the same page.  Looks like a bug, so thank you very much for reporting it.  Not sure what's up yet, but it seems to be extremely repeatable, as the issue occurs on every layer.  That's what I get when I go messing around with the core too much :)

FormerLurker, 2021-02-04:
OK, this is a bit crazy.  I can't believe I've never seen this before, but the program is not handling empty lines that fall between arcs segments correctly.  I must have removed something critical in the past.  Hopefully I'll have time to dig in and fix ASAP.

FormerLurker, 2021-02-04:
You might try [the latest artifacts](https://github.com/FormerLurker/ArcWelderLib/actions/runs/538259176).  Warning:  the MacOs build will not work, and I have not created a raspberry pi build yet.  

I'm having trouble getting the mac build working from github actions, and github has no runner for raspbian yet, but hopefully I'll work that out.  Let me know if it is a problem.

gkatev, 2021-02-05:
I pulled the devel branch, I suppose it is the same as the artifacts. I'm not dekstop Linux, so no build problems!
As far as I can tell the problem is fixed; model looks fine after welding, and I can see a reduction of G1 commands (14269->7756), as well as an increase in G2/G3 (76/215).

Thanks for the fix and for all your work with this project!

FormerLurker, 2021-02-05:
Oh, I'm relieved.  That was the worst bug I've seen in months.  I'm really glad you pointed it out.  I'm going to call this closed, and will try to push out a new release ASAP.

closed 2021-02-05 by FormerLurker