FormerLurker/ArcWelderLib #36: No heatup (build plate and nozzle) when used with Cura and Ultimaker 2+

Invzblio, 2021-02-05:
Generated gcode with Cura and ran it through AW command-line.
When I load up the file in the UM2+, it starts printing right away without heating up the build plate and nozzle. The original file doesn't have this problem. 
I tried with the Cura plugin and got the same results.
I ran the same STL in Simplify3D and ran AW on the gcode. Everything worked fine so it seems the issue is really with the gcode coming out of Cura.

Both gcode files are available in the aw.zip file. 
Original file: UM2_high_side.gcode
Arc Welded file : high_side.aw.gcode

[aw.zip](https://github.com/FormerLurker/ArcWelderLib/files/5935038/aw.zip)

FormerLurker, 2021-02-05:
Ok, so Ultimaker's are a bit odd.  I've never owned one, and just learned of this, but the first line of any ultimaker gcode apparently needs this comment:

;FLAVOR:UltiGCode

Else it just won't work.  This is VERY non-standard behavior.  Arc Welder (like many other pre-processors and slicers) adds a header to the gcode.  This pushes the ultimaker comment down, and causes problems.

Try this:  in your high_side.aw.gcode file, add ```;FLAVOR:UltiGCode``` to the very first line (can't be the second, MUST be the very first line) and see what happens.

This is why we can't have nice things, lol!  Anyway, I'll see what I can do to prevent this, but I really wish ultimaker had chosen a different way to handle this.

Invzblio, 2021-02-06:
@FormerLurker You are right. Moving the line up solves the problem. However, in my Simplify3d gcode file, this line is not present at all. So it seems to be the combination UM2+ and Cura that generates this. Anyways, thanks for looking into it. Are you planning on fixing the issue in the code or I'll have to manually do it for the near future?

FormerLurker, 2021-02-06:
It is the gcode flavor setting that does this.  Give me 30 min and i will push an update.  The faster you can test it the better, since my inbox is blowing up today, 😂

FormerLurker, 2021-02-06:
Ok, try the [new artifacts here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/542027789).

Note that the macos version doesn't work when built from the Github Runners for some reason (still working on that).  Hopefully that's not the one you want to use.

This fix moves that single comment to the top.  I'm REALLY hoping I don't have to move the rest of them up too 🤞. That would get very ugly.  As it is this fix was pretty easy.

FormerLurker, 2021-02-06:
OH, I also pushed a commit to the OctoPrint plugin, which is available on the devel branch if you need that as well.

Invzblio, 2021-02-06:
Works great! Thanks for the quick fix! Great work you did there! I'll be using this tool really often! :)

closed 2021-02-06 by Invzblio

FormerLurker, 2021-02-06:
Thanks, enjoy, and tell your friends, lol

seantapscott, 2021-02-07:
Hey, I was reading through a couple of these.  Wanted to offer some information on this, as I have a bit more experience with Ultimakers.  

This `;FLAVOR:UltiGCode` must be an UM2 problem, as UM3's gcode files state Griffin as the flavor. I think it is actually rather strange that the UM2 is even working with arcs, as UM3 refuses to function with them. 

Here's a UM3 header:
`;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:4.6.2
;GENERATOR.BUILD_DATE:2020-07-13
;TARGET_MACHINE.NAME:Ultimaker 3 Extended
;EXTRUDER_TRAIN.1.INITIAL_TEMPERATURE:200
;EXTRUDER_TRAIN.1.MATERIAL.VOLUME_USED:25211
;EXTRUDER_TRAIN.1.MATERIAL.GUID:506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9
;EXTRUDER_TRAIN.1.NOZZLE.DIAMETER:0.8
;EXTRUDER_TRAIN.1.NOZZLE.NAME:AA 0.8
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:55
;PRINT.TIME:10412
;PRINT.GROUPS:1
;PRINT.SIZE.MIN.X:41.287
;PRINT.SIZE.MIN.Y:6
;PRINT.SIZE.MIN.Z:0.27
;PRINT.SIZE.MAX.X:204
;PRINT.SIZE.MAX.Y:164.722
;PRINT.SIZE.MAX.Z:9.87
;END_OF_HEADER
;Generated with Cura_SteamEngine 4.6.2`

FormerLurker, 2021-02-07:
If UM3s don't support arca, i don't think we need to do much with the Griffin header.  I wonder why G2/G3 is removed?  

Also, just out of curiosity, does the start of header need to be on the first line?  Also, where can I find info on what all of these comments do?  I did a bit of google fu, but sadly my skills were lacking.

It seems odd that comments were used for control rather than adding/adjusting a gcode for this purpose.  I sort of wish i had one of these to play around with so I could understand the philosophy a bit better.

Thanks for your comments!!

seantapscott, 2021-02-07:
No, I am also confused about the UM2 supporting arcs and the UM3 not supporting arcs, which I was mentioning it.  Trying to point out that you probably don't want to make generalized assumptions about Ultimakers, as each model line seems to have different capabilities.  

The UM3 is largely abandoned, doesn't support linear advance, or arcs, or much of anything else.  

As to the header file itself, there are some requirements about what is supposed to be in it, but I'm not entirely sure what that is, but it used to cause problems when slicing with slicers other than Cura. I believe a lot of this is used in their cloud software when trying to determine which printer in a printer workgroup is able to print a specified file, since there are different nozzle configurations (0.4 vs 0.6 vs 0.8), and it tries to match the slicing file to the different printer configurations.

FormerLurker, 2021-02-07:
Thanks for that.  Can i assume that another gcode flavor could be used, and that standard warmup commands, etc, would work normally?

seantapscott, 2021-02-07:
No idea, honestly. I don't do a lot of fiddling outside of Cura with the Ultimaker 3.  Most of what I want to do can be done within the infrastructure that UM has created. 

I had talked with you previously about another issue where I had arcWelder post-processed a file, and it had pushed down the header, but the issues were with missing retractions, and a ton of other artifacts.  People on the Facebook Ultimaker community group stated that arc support is basically not there, so it is tough to say why that happened.