FormerLurker/ArcWelderLib #43: Feedback for V1.1.1

FormerLurker, 2021-02-08:
Please let me know if you've used the [new build of ArcWelder](https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.1.1) or any of the associated libraries. I would love to know what you like, what you don't, what you love, and what you hate. If you have discovered any bugs, please mention those here. I may ask you to create a separate issue for any newly discovered bugs.

Thank you all so much! Without you all I couldn't have done this, and I'm feeling really good about the core algorithm now.

fieldOfView, 2021-02-08:
Using [dependencies](https://github.com/lucasg/Dependencies), it looks like the win32 binaries are not statically linked (ie: they require runtime libraries being installed).

FormerLurker, 2021-02-08:
@fieldOfView, wow, I haven't heard of that exe.  Checked it out, and found exactly what you did.  Will push up a statically linked version ASAP.  Hopefully I will be able to get the github runner to create an X86 version (and the mac version) automatically.  FYI, I'm trying to remove the double zip issue (it's apparently a well known thing), and clean up the file names too.

fieldOfView, 2021-02-08:
So you write a brilliant postprocessor to solve a major problem people are having, and you get to spend your time on setting up build processes.... I feel for you.

DBa2016, 2021-02-25:
Using your Win64 build for a post-processor in PrusaSlicer, works really nice. The output is a bit messy, especially on large files you cannot tell at a glance the current progress, but that's not a major issue.

FormerLurker, 2021-02-25:
@DBa2016, are you printing from prusa slicer, octoprint, sd?  Depending on your printer, PrusaSlicer can output progress gcodes.  You have to enable expert mode:

[image]

Then you should enable the feature:

[image]

If you are using OctoPrint, try Print Time Genius.  It can accurately predict how long a print will take.

Thank you for your feedback!  I'm pretty convinced this release is stable enough for release now.  Going to hopefully drop it this weekend if I can find the time!

DBa2016, 2021-02-25:
Nono, I mean, progress of ArcWelder itself - I am running a batch script which converts the file (print settings->output options->post-processing scripts).
A typical log line from ArcWelder.exe looks like follows:
````
Progress:  percent_complete:30.24, seconds_elapsed:5.02, seconds_remaining:11.57, gcodes_processed: 762594, current_file_line: 786000, points_compressed: 429171, arcs_created: 75709, num_firmware_compensations: 0, compression_ratio: 1.69, size_reduction: 41.00% 
````
"30.24" is the most important information here ("where are we with the conversion? how long will it take to convert?"), however it is not visually identifiable at one glance. Yes, it is there, but then again it is somewhat hidden, not standing by itself... You know? 
It is not a big deal, but I think it would improve the usability to have it more clear... For example, the above log line could maybe look like follows:
````
Progress:  30.24%    5.02/12 seconds, gcodes_processed: 762594, current_file_line: 786000, points_compressed: 429171, arcs_created: 75709, num_firmware_compensations: 0, compression_ratio: 1.69, size_reduction: 41.00% 
````
Something like that. Or even without leading "Progress:".

FormerLurker, 2021-02-25:
> Having said that, this software is still awesome and I am very looking forward to test it tomorrow on a vase which fails miserably without AW :)

Make sure you enable 3D arcs, else it won't work :)

I've had the same issue with the progress.  The output can be parsed by other programs that use ArcWelder, so I'll have to be careful not to break the ones that are out there already.  I could even add some additional parameters that would create a progress bar or something.  Will look into it.

lunartique, 2021-03-18:
I found the 1.1.1 lib will generate random small holes during print (may be because extrusion)
While the same does not for 1.1.0 lib.
[image]

FormerLurker, 2021-03-18:
Interesting @lunartique, can you create a separate issue and post the before/after gcode please?

lunartique, 2021-03-20:
> Interesting @lunartique, can you create a separate issue and post the before/after gcode please?

@FormerLurker are there a reliable way to upload gcode files? it crash my browser everytime i post a gcode.

FormerLurker, 2021-03-20:
Try gist.github.com and paste a link here.

lunartique, 2021-03-20:
> Try gist.github.com and paste a link here.

ok, issue up
https://github.com/FormerLurker/ArcWelderLib/issues/62

FormerLurker, 2021-03-20:
Thanks, will take a look!

Serox13, 2023-07-19:
I have been using arc welder for about 3 months in cura and as post processing in SuperSlicer. 95% on windows the other 5% linux
Cura has exaggerated times when arc welder is used. I put the same settings and same 30x30mm cube cura had it at 35mins and superslicer had it over an hour. I have printed both files and they both take about an hour to print and both look identical. Not sure if this is a Cura issue trying to make print times look good. Either way prints times are taking with a grain of salt and are just roundabout figures
After post processing a black screen pops with some information its to fast to read but the glimpse i caught i think it has the same info as the start GCODE when viewed in Vis studio. Is there anyway to slow this down to actually view?

Besides that it runs smoothly, most files i have run through arc compress about 90% really doesn't make much difference when klipper is installed. Before klipper i found arc welder to be hugely beneficial in running the machine better, maybe i haven't run large enough files through it to see any difference with klipper. 

I have noticed smoother finishes pre/post klipper PLA prints are almosr flawless on the outside, and a big difference in performance pre klipper

Either way thank you for making arc welder it is very useful and exactly what the community needed.