FormerLurker/ArcWelderLib #44: Marlin Bugfix Introduces new logic for arcs

yash-fn, 2021-02-08:
Not an urgent fix, but thought I'd just give you a heads up that Marlin firmware bugfix branch had implemented new arc logic sometime last year. 

This is the part of your code that differs from that version:
https://github.com/FormerLurker/ArcWelderLib/blob/efb6c54e9f07edb25304a35e58431fdfe321c705/ArcWelderInverseProcessor/inverse_processor.cpp#L266-L323

The equivalent section in marlin branch is lines 81-115 from the G2_G3.cpp file within gcode/motion:
https://github.com/MarlinFirmware/Marlin/blob/e7c711996bd3080f5e343eff5556736cbf2e2416/Marlin/src/gcode/motion/G2_G3.cpp#L79-L110

Basically, they fixed a bug involving precision issues when doing full arcs and also fixed another issue whereby G2 could have more or less segments than an equivalent G3 if over 180 degrees.

FormerLurker, 2021-02-08:
Interesting, thank you for posting that.  I was aware of one of those bits, but not the other.  I just started adding the capability to use different interpolation functions, so this is a good use case.  I just need to go through the history for that file and implement any significant changes.  I like the idea of having all variations since I deal with users who are using the bleeding edge development code and users who are using 1.0.

yash-fn, 2021-02-08:
Sounds great! Also not really an issue until they push it into the release branch. Until then technically its "bugfix". 
I've got some ideas myself that I think will help you, but I will have to first figure out Marlin, which is rather quite complex due to the variety of boards they have to support and which is how I came across this code difference in the first place. 
Should I close this issue? Or leave it open?

FormerLurker, 2021-02-08:
Leave it open.  I will add the changes and you can try it out.

yash-fn, 2021-02-08:
Roger roger

closed 2021-09-18 by yash-fn