FormerLurker/ArcWelderLib #49: ArcWelderStraightener seems to assume absolute extruder values

orjanekeberg, 2021-02-18:
Hi!
Using current version (master, efb6c54e9), I tested to convert a small gcode-file into using arcs, and then back again to replace the arcs by lines. The original gcode uses relative extrusion distances, and the arc-version seems to be correct. However, in the "straightened" gcode, all G1 commands corresponding to former arcs have progressively increasing E-values.

Seems like the ArcWelderStraightener has not picked up that the E-values are relative. The gcode I tested is from PrusaSlicer and uses M83 to switch to relative E-values.

FormerLurker, 2021-02-18:
Yeah, that doesn't sound surprising.  I haven't ever printed from the Straightener, and mostly use it to figure out what the firmware is doing.  It should be pretty simple to get it working as expected, though, so i'll see what I can do.  Thanks for letting me know!

FormerLurker, 2021-02-18:
Ok, try [these artifacts](https://github.com/FormerLurker/ArcWelderLib/actions/runs/579675901)(scroll to the bottom).  The MacOs build likely doesn't work.  Can't figure out how to make it work in the actions for some reason...

Also, I haven't tested this much, but I figured since it is broken anyway, I might as well let you take a look.  I also prevented feedrate output if the feedrate hasn't changed from the previous command, and added any comments from the G2/G3 files to the output as well (though that will add a bit of size).

Let me know how it goes!

orjanekeberg, 2021-02-18:
I did a quick test, and it does look better, but still not correct. The E-values for each sequence of G1-commands that come from a straightened arc are still absolute (locally, within that sequence). For example: consider an arc replaced by, say, three lines. The first line has the correct E-value, but the second has a too large value (sum of extrusion for first and second line). Third line has an even larger E-value (same as for the entire arc). Hope this makes sense.

FormerLurker, 2021-02-18:
Ah, yes.  I did the edits too quick hoping i could solve it still today.  It is going to be more challenging than my first attempt.  I should have caught that, thank you for looking!

FormerLurker, 2021-03-28:
Ok, so this took me longer than I'd hoped, but I think I've got it working now.  See [this commit](https://github.com/FormerLurker/ArcWelderLib/archive/65768d59a9ed785ddb740fc6df67d2912d2a5bdf.zip), and please let me know if you notice anything odd.

orjanekeberg, 2021-03-30:
It looks correct now, at least for my test case.

closed 2021-03-30 by orjanekeberg

FormerLurker, 2021-03-30:
Awesome!  Reopen this issue if you encounter any problems.