FormerLurker/ArcWelderLib #50: Feature Request - ArcWelderConsole parameter for log output filename

k7beb, 2021-02-20:
Low priority but I would love to have a parameter in ArcWelderConsole that allows us to direct log output to a named file instead of stdout. I use the ArcWelder.exe postprocessor with PrusaSlicer and Prusa does not provide any way (that I've found) to capture the console output. The window opens and closes before I can see the log. I'd like to be able to see what was accomplished.

FormerLurker, 2021-02-21:
What OS are you using?  You can log console output to a file pretty easily, but it depends on the OS.  For example, for windows use for example:
```
c:\ArcWelder.exe output.gcode >> arcwelder.log
```
The ```>> filename.extension``` at the end forces console output to a file, and appends any file that is already there.

FormerLurker, 2021-02-21:
@k7beb, I realize now you're using PrusaSlicer (I should read and re-read these issues, geesh...), so what I said won't work since the target file is always added as the final parameter.  Bummer, but you could still use the console application.

Is there anything in particular you're looking for?  Statistics??  Something else?  There may be a different approach.  Messing with console logging could be a challenge, but I'll try to think of a way to do it without messing up the code too much.  There is a LOT of logging in there.

k7beb, 2021-02-21:
Thanks for following up. Yes, I'd already tried redirecting output and found out that PrusaSlicer always tacks the input filename on the end so it didn't work. 

Really, I just wanted to be able to see the statistics - amount of compression, number of segments eliminated, etc. that I can see when I use it as a plugin. 

I had been hoping it was easy to just add a parameter and redirect the log output to a text file, but if that's not the case I can certainly just run the postprocessing manually and see what I need. 

Thanks for taking the time to respond. :)

k7beb, 2021-02-21:
Actually, with a little further thought, I can replace the direct call to ArcWelder.exe with a small script and do what I want. Thanks for making me think a little harder about this.

closed 2021-02-21 by k7beb

FormerLurker, 2021-02-21:
Hey, maybe share your script if you can generalize it?  I did do a bit of reading about redirecting via code, so maybe i will take a stab at it at some point.

k7beb, 2021-02-21:
The script I made is very basic and a little ugly, as it leaves a temporary output file named awlog.tmp lying around for Windows to clean up.
I just created a text file named ArcWelder.bat with the following content:

TITLE Arc Welder Pre-processor
<path>\ArcWelder.exe %1 %2 > awlog.tmp
START notepad awlog.tmp
EXIT

To use it modify the third line to point to ArcWelder.exe on your system. Note that you can also add any parameters you want to that line before the %1. The %2 is there in case PrusaSlicer some day decides to add an output file name to the call. Currently it does nothing. 

In PrusaSlicer (or whatever slicer) point to the ArcWelder.bat file instead of ArcWelder.exe to run this script.

All this does is redirect the output from ArcWelder to a file named awlog.txt (it will be overwritten if it already exists) and opens a new Notepad window with that file. 

I had initially thought I could just use a PAUSE so I could look at the output before hitting Enter to continue, but it turns out PrusaSlicer doesn't like if if the script doesn't exit immediately, so I had to go this way instead.

FormerLurker, 2021-02-21:
That's a pretty reasonable way to do it.  As soon as I have some time I'll look into redirecting output from within the code and will post back.  I'll share your script with anyone else who wants this.