FormerLurker/ArcWelderLib #51: Less arcs welded than expected on simple piece

atsju, 2021-02-20:
Hi, 
I use the arc welder plugin in Cura and well I did a simple round piece, it's a single wall 100mm diameter ring.
I basically wanted to see if this improved the finish on arcs with Marlin 2.0 and my printer.
The piece is basically a circle so I expected almost everything to be replaced by curves but it's not the case. I suppose this come from the pieces STL export or from my settings. Can you help me configuring this ?
[circle.zip](https://github.com/FormerLurker/ArcWelderLib/files/6015266/circle.zip)
[CFFFP_circle.gcode.txt](https://github.com/FormerLurker/ArcWelderLib/files/6015261/CFFFP_circle.gcode.txt)
[CFFFP_circle_arc.gcode.txt](https://github.com/FormerLurker/ArcWelderLib/files/6015262/CFFFP_circle_arc.gcode.txt)

FormerLurker, 2021-02-21:
Hey there!  I'll take a look at this ASAP.  I'm sure it's something simple.

FormerLurker, 2021-02-21:
OK, the gcode isn't high enough resolution to work with the default.  The segments are quite large, and the default resolution for arcwelder is very high (0.05mm).  You can either set Arc Welder resolution to a higher value or increase the resolution of your gcode or model.  

For the easiest option, just set the resolution in ArcWelder to 0.1.  I recommend you set it back to 0.05 when you are done.

atsju, 2021-02-21:
@FormerLurker Thank you I tried exporting more precise STL and increasing resolution to 0.075mm and both worked well.

Note that in Cura the setting is limited to 0.025-0.075mm range. But this may be limited by @fieldOfView. I let the ticket open just so you can validate together if the range in Cura is OK and not too restrictive.