FormerLurker/ArcWelderLib #52: Prusaslicer Viewer 2.x doesnt recognize files processed by ArcWelder

gdeconto, 2021-02-21:
This is more of an fyi or very minor issue, but the PrusaSlicer 2.x viewer expects a line that looks something like this at the top of the file:

; generated by PrusaSlicer 2.3.0+win64 on 2021-02-21 at 22:41:55 UTC

ArcWelder puts its own info (for example) at the top:

; Postprocessed by [ArcWelder](https://github.com/FormerLurker/ArcWelderLib)
; Copyright(C) 2020 - Brad Hochgesang
; resolution=0.05mm
; path_tolerance=5%
; max_radius=1000000.00mm
; default_xyz_precision=3
; default_e_precision=5

Even though they are just comments, this results in the viewer producing a popup error message stating:

"Internal error: Not a PrusaSlicer / Slic3r PE generated g-code" 

It is a simple cut and paste fix for users, of course, but thought you should know.

Maybe the comments could be appended to the gcode file instead of prepending?

FormerLurker, 2021-02-21:
I can just detect that comment and add it as the first line.  However, that doesn't work for ultigcode, and since they wont print right without the first line comment, that will take precedence.  Consider posting an issue to the PrusaSlicer github askint them to search the first N lines (maybe 100 or 1000) for their comments.  If arcwelder can do this (and it does), PrusaSlicer can too.

FormerLurker, 2021-02-21:
Also, just FYI, prusaslicer can't display g2/g3 commands.

gdeconto, 2021-02-23:
there is an issue logged in github; https://github.com/FormerLurker/ArcWelderLib/issues/52#issuecomment-782949189

gdeconto, 2021-02-23:
> Also, just FYI, prusaslicer can't display g2/g3 commands.

there is an issue logged in github for this; https://github.com/prusa3d/PrusaSlicer/issues/4352

gdeconto, 2021-02-23:
added new issue in github for the PrusaSlicer exception error issue; https://github.com/prusa3d/PrusaSlicer/issues/6117

FormerLurker, 2021-02-23:
Thanks for adding that issue!  I'll try to tackle it on this end as well.

Also, regarding the issue you linked to above (4352), that is for adding arc generation to PrusaSlicer.  I'm guessing they would fix visualization too in that case, but one could fix the rendering issue before adding native arc support.

FormerLurker, 2021-03-28:
So, I tried forcing the prusa slicer comment to be the first line, but the viewer crashed.  I think it just isn't capable of interpreting G2/G3 :(

Might be time to submit an issue on [Prusa's github](https://github.com/prusa3d/PrusaSlicer/issues).

gdeconto, 2021-03-28:
> So, I tried forcing the prusa slicer comment to be the first line, but the viewer crashed. I think it just isn't capable of interpreting G2/G3 :(
> 
> Might be time to submit an issue on [Prusa's github](https://github.com/prusa3d/PrusaSlicer/issues).

already tried that myself with same results.  looks like there is a commit that may fix this:

"enricoturri1966 added a commit that referenced this issue on Feb 24
@enricoturri1966
#6117 - Allow import of gcode files processed by ArcWelder"