FormerLurker/ArcWelderLib #53: Install problem in octoprint (docker)

melwinek, 2021-02-22:
```
Installing plugin "Arc Welder" from https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip...
/usr/local/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpo8ts61zl/ArcWelderPlugin-master.zip --no-cache-dir
Processing /tmp/tmpo8ts61zl/ArcWelderPlugin-master.zip
Requirement already satisfied: six in /usr/local/lib/python3.8/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.15.0)
Requirement already satisfied: OctoPrint>1.3.8 in /usr/local/lib/python3.8/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.5.3)
Requirement already satisfied: setuptools>=6.0 in /usr/local/lib/python3.8/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (53.0.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.0)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.4.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.12.1)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.1)
Requirement already satisfied: netifaces<1,>=0.10.9 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.9)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2020.8.7)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.5.0)
Requirement already satisfied: psutil<6,>=5.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.8.0)
Requirement already satisfied: flask<2,>=1.1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.2)
Requirement already satisfied: rsa==4.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (4.0)
Requirement already satisfied: feedparser<7,>=6.0.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (6.0.2)
Requirement already satisfied: watchdog==0.10.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.4)
Requirement already satisfied: blinker<2,>=1.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.4)
Requirement already satisfied: netaddr<1,>=0.7.19 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.8.0)
Requirement already satisfied: markdown<3.2,>=3.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.1.1)
Requirement already satisfied: frozendict<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.2)
Requirement already satisfied: zeroconf<0.25,>=0.24 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.24.5)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.0)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.4.21)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.0)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.11.3)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.8.5)
Requirement already satisfied: Click<8,>=7.1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (7.1.2)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.7)
Requirement already satisfied: future<1,>=0.18.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.18.2)
Requirement already satisfied: websocket-client<1,>=0.57 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.57.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.7.0)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.20.3)
Requirement already satisfied: tornado==5.1.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.1.1)
Requirement already satisfied: cachelib<1,>=0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.1)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2021.2.4)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.5)
Requirement already satisfied: requests<3,>=2.23.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.25.1)
Requirement already satisfied: regex!=2018.11.6 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2020.11.13)
Requirement already satisfied: pylru<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.2.0)
Requirement already satisfied: sarge==0.1.5post0 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.5.post0)
Requirement already satisfied: emoji<1,>=0.5.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.6.0)
Requirement already satisfied: pyasn1>=0.1.3 in /usr/local/lib/python3.8/site-packages (from rsa==4.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.4.8)
Requirement already satisfied: pathtools>=0.1.1 in /usr/local/lib/python3.8/site-packages (from watchdog==0.10.4->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.2)
Requirement already satisfied: sgmllib3k in /usr/local/lib/python3.8/site-packages (from feedparser<7,>=6.0.2->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.0)
Requirement already satisfied: webassets>=2.0 in /usr/local/lib/python3.8/site-packages (from Flask-Assets<3,>=2.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.0)
Requirement already satisfied: pytz in /usr/local/lib/python3.8/site-packages (from Flask-Babel<2,>=1.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2021.1)
Requirement already satisfied: Babel>=2.3 in /usr/local/lib/python3.8/site-packages (from Flask-Babel<2,>=1.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.9.0)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/local/lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.26.3)
Requirement already satisfied: chardet<5,>=3.0.2 in /usr/local/lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (4.0.0)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2020.12.5)
Requirement already satisfied: idna<3,>=2.5 in /usr/local/lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.10)
Requirement already satisfied: ifaddr in /usr/local/lib/python3.8/site-packages (from zeroconf<0.25,>=0.24->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.7)
Building wheels for collected packages: Arc-Welder
Building wheel for Arc-Welder (setup.py): started
ERROR: Command errored out with exit status 1:
[... cut ...] d().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-x_5096ew/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /octoprint/plugins/include/python3.8/Arc-Welder Check the logs for full command output.
Running setup.py install for Arc-Welder: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

FormerLurker, 2021-02-22:
It looks like the build tools aren't installed:
```
unable to execute 'gcc': No such file or directory
```
Try installing it from the terminal like so:

```
sudo apt-get install build-essential
```

melwinek, 2021-02-22:
Thank you. Now it is working.
I wonder why it is so, it is the original octoprint container from docker.io.