FormerLurker/ArcWelderLib #55: ArcWelder slows down a lot on certain parts.

illusionmanager, 2021-02-25:
For my Snapmaker 2.0 I'm using Cura. I made a tiny cylinder but whenever it is printing the gyroid type infills, part of  of it is printed at a crawling speed. 
Here is the video:
https://user-images.githubusercontent.com/79637793/109184664-85938b80-778f-11eb-8abd-f8253616c4a2.mp4
While writing this down, I just realized that the only difference between the infill and the rest of the cylinder is that it contains G2 commands. Perhaps there is a firmware issue related to G2.
I'll try to check on that and report later on what I find.

I tried version 1.1.0 and version 1.1.1 both with the same result.
I've included the gcode files here (don't know how to send them otherwise yet), but had to rename them to .txt files in order to do so.
[no-arc-welder.gcode.txt](https://github.com/FormerLurker/ArcWelderLib/files/6044357/no-arc-welder.gcode.txt)
[arc-welder-1.1.0.gcode.txt](https://github.com/FormerLurker/ArcWelderLib/files/6044358/arc-welder-1.1.0.gcode.txt)
[arc-welder-1.1.1.gcode.txt](https://github.com/FormerLurker/ArcWelderLib/files/6044359/arc-welder-1.1.1.gcode.txt)

FormerLurker, 2021-02-25:
Thanks!

> I've included the gcode files here (don't know how to send them otherwise yet), but had to rename them to .txt files in order to do so.

The way you did it is fine.  You can always use gist.github.com, and you can upload multiple files, leave comments, etc.  It's pretty convenient.  Looks like you're getting the hang of github :)

I'll try to take a look at this tomorrow.  Unfortunately, this kind of problem takes a LOT of time to review.  I DO have a suspicion though...  If it's not a problem with the gcode (which it may well be), it MIGHT be that the feedrate for G2/G3 isn't being applied.  If that's the case we will be able to test it.

That being said, I'm just speculating.  Will know more once I review all of the files.

illusionmanager, 2021-02-25:
I just did send both G2 and G3 commands directly from a console to the Snapmaker printer. The printer does make clockwise and anti clockwise circles at full speed.
I also tried this sequence
G2 X40 I40 F300
G3 X40 I40 F3000
G2 X40 I40
just to make sure the the G2 command also picks up the F3000 from the previous G3 command, and it does.
So it seems both do work just fine on the Snapmaker 2.0.

FormerLurker, 2021-02-25:
Ok, thanks for running those tests!

stevenhu70, 2021-08-28:
Firstly - I'm a relative noob when it comes to more advanced slicing. I do some designing (F360) and printing (CR-10s Pro V2). I have never had problems I can't work out before... however a recent print using some very large (approx 20cm diameter circular arcs) was causing zits.

I was ecstatic when I found ArcWelder and immediately installed the plugins for Cura and OctoPrint.

My first test was amazing - (on a small print with the same issue) ArcWelder totally fixed the problem and the print was amazing.

I'm finally getting to the point, I promise...

The large objects processed fine through ArcWelder, however, at several points the printer slows to a crawl (much the same as illusionmanager's video. As I type I'm allowing the 12+ hour print to continue and, so far, it looks great, but the slowing is causing significant delays. I'll be happy to send the gcode files (pre and post processed) however they are 12MB and 4MB respectively.

So I was just posting here to say that I believe I have the same problem and would love to see a fix.

Thanks for the work you've done on ArcWelder - so far (on smaller models) it's been fantastic!

STL's attached for illustration purposes :)

[Fan Cowling Front aqnd Back.zip](https://github.com/FormerLurker/ArcWelderLib/files/7069909/Fan.Cowling.Front.aqnd.Back.zip)

FormerLurker, 2021-08-30:
@stevenhu70, This is likely the result of some firmware issues.  Can you post your firmware version?

stevenhu70, 2021-08-30:
OK... so I managed to install the Octopi plugin version v1.1.0rc2 and can now see the firmware compatibility (this had me confused for a little while). Now I can see that the firmware checker believes there is a problem:

Name:Marlin 
Version:2.0.5_DW6.2
Supported:No
Arcs Supported:Yes
3D-Arcs Supported:Unknown
Arcs Enabled:Yes
Warnings:
Your firmware version indicates that it is not supported, but arcs appear to be supported and enabled. Use with caution.
Cannot determine if 3D Arc commands are supported (for use with vase mode). Since 3D arcs are currently disabled, this should be OK.
Last Check On:08/30/21 21:50:21

I believe that this is the TinyMachines flavour of Marlin, but I haven't done a firmware upgrade for over a year. I guess it's time to  re-learn how to do that again.

I also realised that this "issue" should probably be in the ArcWelderPlugin repository not in the ArcWelderLib. My bad, wasn't paying attention. I'll post again when/if I manage to update my firmware 😢 

Many thanks for responding so quickly!
Steve

FormerLurker, 2021-08-30:
You can leave it here.  It doesn't really matter to me :)

I would recommend you update the firmware.  Starting with 2.0.6, arc support was improved a lot.  Specifically I believe the slowdowns could be caused by the junction deviation algorithm (not 100% sure there), and I think that was addressed in a newer version.  Anyway, it's always nice to have the latest firmware goodies :)

stevenhu70, 2021-08-30:
Nailed it. The TinyMachines firmware based on 2.0.9 seems to have fixed the issue I was having.  I'm going to do a test print overnight tonight just to confirm... but it seems to be good (based on two layers of a model I was printing that exhibited the problem.

As a bonus the auto bed levelling (BLTouch) with the new firmware is a lot faster. Winner winner chicken dinner!

New report from ArcWelder is:
Name:Marlin 
Version:2.0.9_DW7.4
Supported:No
Arcs Supported:Yes
3D-Arcs Supported:Unknown
Arcs Enabled:Yes
Warnings:
Your firmware version indicates that it is not supported, but arcs appear to be supported and enabled. Use with caution.
Cannot determine if 3D Arc commands are supported (for use with vase mode). Since 3D arcs are currently disabled, this should be OK.
Last Check On:08/30/21 23:26:33

THANK YOU!

FormerLurker, 2021-08-30:
Of course!!! I may ask you to send a gcode to your printer to get a response if you don't mind.  Will get to that tomorrow, and I can add it to the firmware list.  

Glad you got this solved, and glad your bed leveling is faster!!!  Bonus :)

stevenhu70, 2021-09-01:
Just let me know if you want me to do anything.... I'll be happy to help.