FormerLurker/ArcWelderLib #56: How to use with PrusaSlicer on Mac

EagleeyeKai, 2021-03-01:
Dear all,
could somebody please help me how to use the script on MacOS version of PrusaSlicer? Meaning, where to save the ArcWelder file and how exactly to insert the path..

Thanks a lot!

FormerLurker, 2021-03-01:
I wish I owned a Mac to test this :(  I will ask around and see if anyone can give me an example, then I can update the guide.

EagleeyeKai, 2021-03-01:
This sounds great. As somebody already updated the MacVersion, I am sure, another user can guide me into the right direction :)

soupkitchenfilms, 2021-06-10:
@EagleeyeKai what is your problem exactly?
I have the ArcWelder script in my PrusaSlicer folder (Original Prusa Drivers)
So the line I added to Post Processing is '/Applications/Original| Prusa\ Drivers/Arcwelder'
The one gotcha is you need to right click and open the ArcWelder executable from the finder the first time so that you can tell the Mac OS to trust the application.