FormerLurker/ArcWelderLib #57: Is this compatible with Cura Arachne

Bob90, 2021-03-03:
Hi,

Is this compatible with Cura Arachne where line widths can vary dynamically?

Thanks,

FormerLurker, 2021-03-03:
I've not tried it, but it should be fine.  I need to see some gcode, but I can't see why it wouldn't.

Bob90, 2021-03-03:
It has variable line width/extrusion.
Does your code check for changes in extrusion values along the arc? Because of it doesn't, you might end up with a single arc with over/underextrusion compared to the variable line width.

Sorry, I'm not a coder so I wouldn't know where to look in your code to see if it does check extrusion.

FormerLurker, 2021-03-03:
@Bob90, good point.  Again, without seeing how the gcode is arranged, it would be difficult to tell, but I do NOT ensure extrusion volume per mm of travel.  It might be interesting to see how close the extrusion volume per mm of travel varies in a typical gcode file without arachne, and see what an acceptable variance is.  It would be trivial to add this to the arc generation constraints, so really all I need to know is what deviation should end any current arcs.

FormerLurker, 2021-03-03:
I installed it, so I should be able to see exactly what's going on.  Thanks for the heads up!

FormerLurker, 2021-03-03:
So, I added rudimentary support for this, but need to test a bit and update the OctoPrint plugin before I push this.  I'm debating adding the extrusion_rate_variance setting as a % instead of a fixed value, which should be pretty easy to set.  I doubt changes of less than 01% (or 0.1) would make much of a difference.  

Here is a typical value for the extrusion rate (mm of extrusion/mm of travel) from a print I just looked at:  0.0313363 

Looks to be +- 0.0002 mm/mm covers almost all extrusions.  That's about 0.64%.  Could do 1% and see what happens.  I'll keep researching this new cura version, and will see what I can find.

FormerLurker, 2021-03-04:
Ok, I added some basic support.  I decided to go with a percent, since that seems to make the most sense, and doesn't require intimate knowledge of the slicer engine to complete.  The new parameter is -v or --extrusion-rate-variance-percent, with a default of 1%.  You can find the latest console app in the artifacts tab (from the devel branch, fyi) if you want to give it a go.  I need to find a good test model for this.  Also, I haven't printed with this build yet, but it is basically just rejecting arcs based on the calculated flow rate variance: ( (flow_rate_current - flow_rate_previous) / flow_rate_current) * 100.0

Bob90, 2021-03-04:
Amazing! I was wondering because I read that Ultimaker will eventually merge this as the default slice engine behaviour. It's still alpha so not immediate, but I did wonder how they would interact.

I'll try, but I'm quite simple and have been using either the OctoPrint plugin or the (normal) Cura add-on.

I'll read how to set parameters.

FormerLurker, 2021-03-04:
I will run a test today or tomorrow too.  I will also post instructions for testing this as soon as i can.

baileyji, 2022-07-05:
As part of closing out this you might consider adding some guidance about what the preferred "G90 Influences Extruder" setting should be here. I've had an very hard time trying to figure out what the fork provenance of this new firmware is.