FormerLurker/ArcWelderLib #58: Need build instructions

TheAndr0id, 2021-03-07:
Unless it's somewhere I forgot to look I couldn't find any build instructions for the source code.  I took a stab and I believe this worked: (I downloaded the source (tar.gz) and from within the extracted directory)

```
mkdir build
cd build
cmake --DCMAKE_INSTALL_PREFIX=/opt/ArcWelder ..
make
make install
```

Did I miss anything?

FormerLurker, 2021-03-07:
Are you wanting to build everything or some specific component?

TheAndr0id, 2021-03-07:
Just looking to build everything.  The final binaries (as I've built them) are less than 1200kbytes, so space isn't an issue.  I just wanted to have a binary under /opt that I can use with PrusaSlicer.

Sorry, not wanting to be critical, there were a few warnings that came up (eg: warning: "_POSIX_C_SOURCE" redefined) but nothing that was all that exciting.  I just wanted to make sure I wasn't missing anything.

FormerLurker, 2021-03-10:
You could just download the binaries [here](https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.1.1) for the latest pre-release.  

Yeah, there are too many warnings, most of which don't affect the functionality but should be addressed.  I'm a bit rusty with c++, as I've only done 2 projects in the last 20+ years :)  I'm getting better though, and I think if I started over things would be nicer, but as it is I've been refactoring things as time permits.

TheAndr0id, 2021-03-10:
No worries, the warnings aren't anything but build noise.  So far the results look good, I have tested it on a STL, but I can't print it yet since my Ender 3v2 doesn't support arcs until I get the firmware rebuilt.

I have a policy that I don't use binaries and build everything from source.  This is half because I don't trust pre-built binaries and half because I use Slackware and many binaries are linked against different library versions.  I just document everything I build so when I do an update it's just a simple copy/paste instead of re-doing the discovery process with each package.

FormerLurker, 2021-03-10:
> I have a policy that I don't use binaries and build everything from source. This is half because I don't trust pre-built binaries and half because I use Slackware and many binaries are linked [...]

Sounds reasonable.  Take a look at the [Github build action](https://github.com/FormerLurker/ArcWelderLib/blob/master/.github/workflows/ccpp.yml).  It contains every step (and then some) necessary to compile the code.