FormerLurker/ArcWelderLib #60: Prusaslicer 2.3.0 path problem

theshortest, 2021-03-11:
I have sliced a few files and I am left with this. I am not able to solve.
[image]

FormerLurker, 2021-03-11:
Please upload the original and the welded gcode to gist.github.com, and drop a link here, thanks!

theshortest, 2021-03-11:
Hope this helps.
https://gist.github.com/theshortest/30373a95d23865f9981a669e852bddd8

FormerLurker, 2021-03-15:
Looks like your visualizer doesn't correctly render arc commands.  Here's how simplify3d sees your welded gcode file:

[image]

So far the only slicer I know that correctly renders arc commands is simplify 3d.  The OctoPrint gcode view and the pretty gcode plugin also do this correctly.  ncviwer.com can mostly do it correctly, though it does have a few issues, but it will give you a general idea.

theshortest, 2021-03-15:
Unfortunately It wasn't just my viewer. It's how the parts are printed. That's how I first found out. Things work fine through Octoprint but I am trying to use PrusaSlicer post processor so I can use Simplyprint.io.

[image]

FormerLurker, 2021-03-15:
Can you try sending a 'G2' command (empty, with no parameters) to your printer via the OctoPrint terminal and see how it responds.  It may disconnect, but it won't cause any problems.  Paste the response in here, thanks!

theshortest, 2021-03-15:
I will after work today. Cannot access my machine at the moment.

theshortest, 2021-03-15:
For what its worth this is how Cura interprets it.

[image]

FormerLurker, 2021-03-15:
There's an issue logged for this at the cura github page.  Hopefully it gets fixed.

theshortest, 2021-03-17:
G2 says "unknown command"

FormerLurker, 2021-03-17:
Arcs are disabled in your firmware. You will need to enable arc commands before you can use arcwelder.

theshortest, 2021-03-26:
Sorry for the delay. I have tried and tried and I still get broken arcs. I cant speak to the arcs being disabled in Marlin but I have successfully printed 6 parts through Octoprint/arcwelder with no firmware changes. All Gcode viewers show broken arcs from Prusasclicer standalone Arcwelder. At this time I do not believe it is viewer problem I believe it is an issue with the standalone executable I'm using with Prusaslicer.

theshortest, 2021-03-26:
Finally had a chance to check myself on simplify and sure enough!
[Screenshot 2021-03-26 134715](https://user-images.githubusercontent.com/7723998/112679210-e61ef280-8e39-11eb-9f21-c56c00cc084b.png)