FormerLurker/ArcWelderLib #61: Improve Marlin 1.1.x ARC Support based on the improvements of Marlin 2.0.6

gustavomassa, 2021-03-19:
Hello, thanks for this amazing lib.

I'm planning to get the Marlin 2.0.6 feature into Marlin 1.1.9 since I use my board is 8bits and I don't want to update to 2.x.
The idea is to compare the differences between the 2x algorithm with the 1.1x algorithm. 
[image]
What you mean by greatly enhanced? What is missing on 1.1x?
Marlin 1.1.x  ARC implementation: [https://github.com/MarlinFirmware/Marlin/blob/1314b31d97bba8cd74c6625c47176d4692f57790/Marlin/Marlin_main.cpp#L14451](https://github.com/MarlinFirmware/Marlin/blob/1314b31d97bba8cd74c6625c47176d4692f57790/Marlin/Marlin_main.cpp#L14451)
Marlin 2.x ARC implementation: [https://github.com/MarlinFirmware/Marlin/blob/b9065195f1babf00ee453c39410206af90a525ae/Marlin/src/gcode/motion/G2_G3.cpp#L25](https://github.com/MarlinFirmware/Marlin/blob/b9065195f1babf00ee453c39410206af90a525ae/Marlin/src/gcode/motion/G2_G3.cpp#L25)

[https://github.com/FormerLurker/ArcWelderLib/issues/18](https://github.com/FormerLurker/ArcWelderLib/issues/18) is also related to it, right?

What do you think about the idea?: I'm newb about ARC Welder, how could I test the modifications? Do you have any STL that I can use to test the output? I'm using Slic3R

FormerLurker, 2021-03-20:
Yes, firmware compensation is basically accounting for the missing min_arc_segments define, which allows accurate geometry for small arcs.  When using only mm_per_arc_segment, small arcs will end up looking like polygons.  Compensation just prevents arcs from being generated if they violate arcwelders min_arc_segments + mm_per_arc_segment setting.

Adding min_arc_segments is the thing you should focus on, though there are other changes in marlin 2.0.6+ that are beneficial, like an enhanced junction deviation algorithm.  You may be able to compensate for that by adding a min_segment_length define to prevent insanely shot segments from being generated.

Why don't you look at my marlin fork of prusa's fork of marlin 1.  I have basically already done most of this there.  It shouldn't be too hard to adapt.  Check my repo list for this.

gustavomassa, 2021-03-20:
@FormerLurker thanks for the info, I will look up your repo as an example

I did the first commit, adding the min_arc_segments, 24 is a good value?
[https://github.com/gustavomassa/Marlin/commit/4f664e8b5398451542b9e1b2e46f3c1d2bccf970](https://github.com/gustavomassa/Marlin/commit/4f664e8b5398451542b9e1b2e46f3c1d2bccf970)

What about the **ARC_SEGMENTS_PER_R** and **ARC_SEGMENTS_PER_SEC**?
I still have to make changes to apply the scaled_fr_mm_s value, this will be complicated, the "scaled_fr_mm_s" depends on the "ARC_SEGMENTS_PER_SEC"

FormerLurker, 2021-03-20:
Those settings are good defaults.  Though the feedrate based settings are cool, they require a bit of testing to properly configure.  Maybe consider implementing those later.

As for a test, the first layer of benchy is pretty good, as it has curvy text.  Slice it with archimedean bottom infill for a proper test.

gustavomassa, 2021-03-20:
@FormerLurker How much infill is needed for the test? "Archimedean Chords" for the bottom only? I cut the benchy file, so it will print the first 2 layers only.

[image]
[image]
[image]
[image]
[image]

**Not sure if I should use the G90/G91 Influence Extruder with my firmware modifications**
This is the result with these 20% infill archimedean settings:
[image]

FormerLurker, 2021-03-20:
Wow, that's some line width :)  

> How much infill is needed for the test? "Archimedean Chords" for the bottom only? I cut the benchy file, so it will print the first 2 layers only.

The infill amount won't batter, because top and bottom infill is solid.  Since you are only printing the first two layers, you won't have sparse infill at all.

> Not sure if I should use the G90/G91 Influence Extruder with my firmware modifications

You should not.

Looks good!  drop a link to your mods.  I'd be curious to see what you have.

FormerLurker, 2021-03-20:
Regarding the link, I pretty easily found your repo, so not necessary.  Nice job!