FormerLurker/ArcWelderLib #64: max_extrude_cross_section error

ilium007, 2021-03-27:
I am running latest Klipper and got the max_extrude_cross_section error. I read on the ArcWelderPlugin issue list a similar issue that was resolved:

https://github.com/FormerLurker/ArcWelderPlugin/issues/99

> There were several issues I fixed, but the primary one causing your issue was the fact that I was not converting the percentages (for example 5%) to a decimal percent (0.05) for the path tolerance percent! I know this was working earlier, so I must have broken it during a recent refactor. Whoops.
> 
> Anyway, before I discovered the path tolerance conversion issue, I added a few more safety mechanisms to prevent the problem you were seeing. The first was arc direction detection change and prevention. Basically if an arc passes all the other safety tests (length tolerance, resolution, etc..), I also check to make sure the direction stays consistent across the whole arc. I think there is actually a limitation with the current implementation of this check that I'm working now to resolve, but I believe the main issue has been corrected.

FormerLurker, 2021-03-28:
I'm still looking into this.  There are a lot of moving pieces, but I believe this has to do with absolute extrusion.  Perhaps you could try re-slicing with relative extrusion (if you were in fact using absolute extrusion) and see if the problem persists.  At least we will have some confirmation in that case.

john-parton, 2023-01-11:
I just got this error, and I changed the "Maximum path deviation" / "Resolution" setting in Cura from the default 0.05 to 0.02 and it worked. I guess it's probably because an arc changes in length too much? I really have no idea

john-parton, 2023-01-11:
I lie. It got part way into the print and failed half way through. Do you want me to upload an STL and the Cura settings I used?