FormerLurker/ArcWelderLib #66: Arc welder creating artifacts in Cura...

PhilBaz, 2021-05-03:
I see the beauty of Arc welder but have had mixed results using it in cura. It mostly works great but has caused problems a few times.

Please see the attached images of 3 prints...the first two are sliced with different setting and using arc welder, the third has Arc welder turned off and the artifact is gone...ill attach the images as well as the G-code (github made me zip it). Is this an issue or do my settings need to change? I'm using settings very close to the defaults.

Thanks for your time and effort!
[image]
[image]
[image]

[G-code.zip](https://github.com/FormerLurker/ArcWelderLib/files/6415531/G-code.zip)

FormerLurker, 2021-05-09:
OK, I did a layer by layer examination of all the over/under extrusions from all the files posted (I had to visually count the layers and compare with the image, so I did +- 1 layer for each) and could find no over/under extrusions in the arc welded gcode that you produced, or my newer version that contains extrusion rate checking.  However, I have been getting a lot of issues regarding your printer in particular, so I suspect a firmware issue.  Please see [this issue](https://github.com/FormerLurker/ArcWelderPlugin/issues/150) in my repo for other users having issues on this particular printer.  I may merge these issues since they seem related.

Also, are you printing via SD or streaming over USB (octoprint, pronterface, etc)?

FormerLurker, 2021-05-09:
OK, you probably saw my post to the cura plugin issues, but I wanted to add a few things:

1. This is definitely a firmware issue.  
2. I was able to eliminate the issue by setting the maximum radius in the arcwelder console app (same in the cura plugin) to 10000mm.
3. I was also able to eliminate the error by setting the N_ARC_CORRECTIONS define to 0 WITHOUT changing any of the arcwelder settings.

Can you give either item 2 or 3 a go?  Preferably both if possible.  FYI, item 2 will reduce the compression a bit (a few percent), but it's worth it if it removes artifacts.  Item 3 may cause firmware slowdowns, fyi, but maybe not.

FormerLurker, 2021-05-09:
@PhilBaz, I created an issue about this on the Marlin 2 github.  Hopefully a firmware solution can be found.  I could potentially also add some additional firmware compensation to arcwelder, though that is not ideal.