FormerLurker/ArcWelderLib #67: Setting Prusaslicer not to overwrite the source file

ssombra, 2021-06-02:
I configured Prusaslicer following the instructions on the page and it is working well. However, I am not sure how to tell Arcwelder not to overwrite the original file but to create a new one.

FormerLurker, 2021-06-02:
Hmm..  I don't think it's possible since prusaslicer doesn't have file name tokens :(  You might want to add a feature request to prusaslicer's github page.  It would be very useful.  Something like {file_name}.gcode, where {file_name} would be replace with the output file name minus the extension.

Just curious why you want this, if you don't mind explaining.

ssombra, 2021-06-02:
Not at all :) Just wanted to see the differences in the files after the Arcwelder processing... pure curiosity, trying to learn more.

ssombra, 2021-06-02:
I guess the workaround is easy, I could just slice two files, one with Arcwelder processing and the other without, but I thought there was a more elegant way. Thanks for the reply and congratulations for Arcwelder, it improved my prints a lot, I was having some blobs when printing through Octoprint.

FormerLurker, 2021-06-02:
> I was having some blobs when printing through Octoprint.

That's exactly why I wrote this!  I didn't have any issues personally, but it depends on your slicer, the model, and your firmware.  I was tired of reading about quality issues using Octoprint, and knew what was causing it, and the fix involves reducing the number of commands/second.  Adding in meatpacker (if your firmware supports it) also helps, and the buffer buddy plugin.  With all 3, slowdowns are a thing of the past unless your printer is ultra fast.

ssombra, 2021-06-03:
I will need to look into meatpacker and the buffer buddy... I don't know what they are, but I am in for anything that helps reducing those issues. Thank you.

FormerLurker, 2021-07-06:
Going to close this issue since it seems to be resolved (more or less).  Thanks for posting!

closed 2021-07-06 by FormerLurker