FormerLurker/ArcWelderLib #69: Strange Gaps and Resultant Seams

tuncof, 2021-06-16:
Hi ya,

To evaluate smoothness of transition between circular and linear regions properly, I did write a simple openscad script like that:

$fn=80;
r=20;
d=10;
module k(){translate([-d,0,0])cylinder(h=10,r=r,center=true);}
rotate([0,0,-20])union(){
k();mirror()k();
cube([2\*d,2\*r,10],center=true);}

Nothing fancy there, just two distanced halves of a cylinder with a height of 10mm and a radius of 20mm, halves are seperated with a distance of 20mm, which is filled with a box filling this very distance with the same height with cylinders. So a simple finger biscuit or a bacteria-like shape. Exporting as a stl file, sliced with Cura 4.4 and then processed with ArcWelderConsole with default settings. Upon investigating gcode in Repetier-Host, there were gaps between circular and linear transition regions. Apparently, these occured to me the culpirit of seamsafter printing, if I'm not mistaken. Tried every other option of arcwelder (including -d option) and even increased segment numberfrom 80 all the way up to 200 ($fn in openscad) to no avail. Strange to me that some regions are perfectly smooth while others are bulgingly seamed. Any ideas?  Thanks.

STL file and gcodes after cura and arcwelder:
[cubecylinder.zip](https://github.com/FormerLurker/ArcWelderLib/files/6663056/cubecylinder.zip)

##
Repetier-Host preview after Cura:
[image]

After Processing with ArcWelderConsole-devel (Greens are perfectly smooth while reds are aberrant after printing):
[image]

Seams:
[image]
[image]

FormerLurker, 2021-06-16:
Can you post the unprocessed and processed gcode please?  Thanks!

FormerLurker, 2021-06-16:
Ahh, nm.  I missed the link in tye email.  Saw it clearly on github, lol

FormerLurker, 2021-07-03:
Sorry this took so long.  I ran both these through a more reliable visualizer for G2/G3, and the gcode paths look identical:

[image]

However, seams can be the result of improper extrusion, which doesn't show up in any visualizer I've ever used.  I did notice this in the header:

```
; arc_welder_g90_influences_extruder = False
```

Are you running marlin 2+ or a fork of marlin 2?  If so, that needs to be set to true.  Try that first if this is the case.

Next, there have been a few updates to the welding algorithm, some of them bugfixes, so I'd also recommend trying the latest dev build (see the actions tab).  Do that after trying the g90/91 influences extruder setting (if it applies to your firmware).  Let me know if you need any help with any of this, and be sure to let me know how it goes!

FormerLurker, 2021-07-03:
Oh, in the previous post, the left image is the unprocessed gcode, and the right image is the arcwelded code.