FormerLurker/ArcWelderLib #70: Processing non printer Gcode files

ithinkido, 2021-06-29:
I would like to process some Gcode for a pen plotter to reduce the code size and hopefully improve smoothness. I have tried with both the Windows and Linux versions, and although it appears that ArcWelder has successfully run, it does not seem to be creating any arc moves.The test file attached should be a good candidate for this.
Is it possible that ArcWelder only works on 3D printer gcode files with a specific format ?
[spiral.nc.txt](https://github.com/FormerLurker/ArcWelderLib/files/6737086/spiral.nc.txt)
[image]

FormerLurker, 2021-07-06:
You are in luck!  I just finished a feature to encode travel moves (moves without any extrusion).  It needs more testing, but here are the results from processing the file you posted:

[image]

Here is a side-by-side comparison of the tool paths:

[image]

The orange lines are G0/G1 commands, the blue lines are G2/G3 commands.  As you can see, much of this file was successfully converted.

If you would like to try out the devel version, let me know and I'll post some instructions.

just-jason, 2021-07-07:
>
> If you would like to try out the devel version, let me know and I'll post some instructions.

Yes please, I definitely would like to try that out.

FormerLurker, 2021-07-07:
Check [here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/1004732650) for the latest bin files.  If you want you can post your gcode and I can do a sanity check before you try it out.

FormerLurker, 2021-07-07:
Actually, i guess you will try the gcode you already posted, which I already reviewed, lol

just-jason, 2021-07-07:
For my use this is PERFECT. The tests I have done are working just as I had hoped. There will be a lot of pen plotter users happy about this. Thanks very much.

FormerLurker, 2021-07-07:
Hooray!  If you have time, could you post a pic of your finished plot?

just-jason, 2021-07-13:
Sorry fro the delay. A strange thing, I went to do a conversion again today, and I can not get Arcwelder to run. Very strange. 

[image]

FormerLurker, 2021-07-13:
Hmm..  I will look at the console app.  It LOOKS like I must have screwed something up.  Try adding a second parameter for giggles, like:

```ArcWelder.exe perlin.gcode perlin.aw.gcode```

and see what happens.  Otherwise I messed up configuring TCLAP for the new parameters.

just-jason, 2021-07-13:
Still the same

FormerLurker, 2021-07-13:
Ok, will do my best to reproduce and fix.  Might take a few days depending.  Thanks!

FormerLurker, 2021-07-16:
Ok, this was totally my fault, lol!  Try [the builds here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/1038801249), which I think should work much better.

just-jason, 2021-07-17:
Looking good now. I was confused at first as the info stats did not show any converted arcs , but the end file has been converted so I would call that a success. 
[image]
[spiral.nc.txt](https://github.com/FormerLurker/ArcWelderLib/files/6834605/spiral.nc.txt)

FormerLurker, 2021-07-17:
I will look at that issue as well. It is a work in progress ☺️

FormerLurker, 2021-07-17:
Ok, [this](https://github.com/FormerLurker/ArcWelderLib/actions/runs/1040527558) should fix the statistics display issue.

just-jason, 2021-07-18:
Perfect !!!
```C:\Users\JASON\Desktop\ArcWelder\bin>ArcWelder.exe spiral.nc
2021-07-18 14:41:11.010 - arc_welder.gcode_conversion - INFO - Processing GCode.
2021-07-18 14:41:11.011 - arc_welder.gcode_conversion - INFO - Arc Welder Arguments
        Source File Path             : spiral.nc
        Target File Path (overwrite) : spiral.nc
        Resolution                   : 0.05mm (+-0.02500mm)
        Path Tolerance               : 5.000%
        Maximum Arc Radius           : 9999mm
        Min Arc Segments             : 0
        MM Per Arc Segment           : 0.000
        Allow 3D Arcs                : False
        Allow Travel Arcs            : True
        Allow Dynamic Precision      : False
        Default XYZ Precision        : 3
        Default E Precision          : 5
        Extrusion Rate Variance %    : 5.000%
        G90/G91 Influences Extruder  : False
        Max Gcode Length             : Unlimited
        Log Level                    : INFO
        Hide Progress Updates        : False
Progress:  00.0% complete - Estimating remaining time.
Progress: 100.0% complete - 0.03 seconds total.
2021-07-18 14:41:11.043 - arc_welder.gcode_conversion - INFO - Target File Travel Statistics:
   Min          Max     Source  Target Change
---------------------------------------------
  0.000mm to   0.002mm       0       0   0.0%
  0.002mm to   0.005mm       0       0   0.0%
  0.005mm to   0.010mm       0       0   0.0%
  0.010mm to   0.050mm       0       0   0.0%
  0.050mm to   0.100mm       0       0   0.0%
  0.100mm to   0.500mm       0       0   0.0%
  0.500mm to   1.000mm     156       9 -94.2%
  1.000mm to   5.000mm    7084    2688 -62.1%
  5.000mm to  10.000mm       0     388    INF
 10.000mm to  20.000mm       0      14    INF
 20.000mm to  50.000mm       0       0   0.0%
 50.000mm to 100.000mm       0       0   0.0%
          >= 100.000mm       1       1   0.0%
---------------------------------------------
Total distance source:............10718.980mm
Total distance target:............10720.180mm
   Total count source:...................7241
   Total count target:...................3100
 Total percent change:.................-57.2%
2021-07-18 14:41:11.044 - arc_welder.gcode_conversion - INFO - Target File Extrusion Statistics:
   Min          Max     Source  Target Change
---------------------------------------------
  0.000mm to   0.002mm       0       0   0.0%
  0.002mm to   0.005mm       0       0   0.0%
  0.005mm to   0.010mm       0       0   0.0%
  0.010mm to   0.050mm       0       0   0.0%
  0.050mm to   0.100mm       0       0   0.0%
  0.100mm to   0.500mm       0       0   0.0%
  0.500mm to   1.000mm       0       0   0.0%
  1.000mm to   5.000mm       0       0   0.0%
  5.000mm to  10.000mm       0       0   0.0%
 10.000mm to  20.000mm       0       0   0.0%
 20.000mm to  50.000mm       0       0   0.0%
 50.000mm to 100.000mm       0       0   0.0%
          >= 100.000mm       0       0   0.0%
---------------------------------------------
       Total distance:................0.000mm
   Total count source:......................0
   Total count target:......................0
 Total percent change:...................0.0%
2021-07-18 14:41:11.046 - arc_welder.gcode_conversion - INFO - Arc Welder process completed successfully.

fabien, 2022-06-07:
I'm trying to use ArcWelder to prepare files for my CNC outfitted with a tangential knife cutting module (my controller handles the knife axis rotation automatically, I just feed it G1/G2/G3 instructions).

Even though my source files don't have any E codes, the G1 instructions are being output as G2/G3 - this is *without* the `--allow-travel-arcs` option (which I think is for non-3D printing use-cases like mine). Granted, it did not work on all files (from other gcode generating programs), so I decided the `--allow-travel-arcs` option instead.

However, when I do set `--allow-travel-arcs` I don't get any G2/G3 conversion at all. So it appears to work exactly in the opposite way for me, which seems very odd. Any thoughts?

FormerLurker, 2022-06-07:
@fabien,

I believe I fixed the inverted option for ```--allow-travel-arcs```, but still need to add detection of the G01 as opposed to G1.  It's unfortunate that these are both valid commands, as this will increase parsing time for everyone :(.  However, compliance is compliance, so I'll get that working too.  Thanks for the report!