FormerLurker/ArcWelderLib #72: `default_xyz_precision` and `default_e_precision` set to 6 if specified

fieldOfView, 2021-07-16:
Specifying the `--default_xyz_precision` or `--default_e_precision` options always results in a value of 6 being used.

In https://github.com/FormerLurker/ArcWelderLib/blob/master/ArcWelderConsole/ArcWelderConsole.cpp#L135 and https://github.com/FormerLurker/ArcWelderLib/blob/master/ArcWelderConsole/ArcWelderConsole.cpp#L141, the `--default_xyz_precision` or `--default_e_precision` options are defined as unsigned char, so they are parsed as (ascii) characters instead of ints. In my understanding this means that specifying a 0 will result in a value of 48 being parsed.

FormerLurker, 2021-07-16:
Will check this out today, thanks for reporting!

FormerLurker, 2021-07-16:
You are 100% correct.  And there were more issues hiding in there.  I just made a ton of changes, and have been testing almost exclusively via the OctoPrint app.  I'm working on some of the other stuff  now, and will post back once this is resolved.

FormerLurker, 2021-07-16:
Ok, I just pushed a fix to devel.  I've got more testing to do, but I think this issue is solved.