FormerLurker/ArcWelderLib #73: PARSE ERROR: Argument: -r (--resolution-mm)  Couldn't find delimiter for this argument! when used in Powershell on Windows 10.

DarkAlchy, 2021-08-07:
Following a video from January and I typed what he typed for the -r command, and it crapped out.

PS E:\Downloads\Windows-MSVC\bin> .\arcwelder.exe -r=.1 body26568.gcode b-aw.gcode
PARSE ERROR: Argument: -r (--resolution-mm)
             Couldn't find delimiter for this argument!

Powershell is the cmd (console) replacement that Microsoft has been heavily trying to get people over to it.

FormerLurker, 2021-08-11:
Hi @DarkAlchy, sorry for the delayed response!  I was out of town last week, and am still playing catch-up!

I am familiar with powershell, but haven't tried it with ArcWelder.  I use a third party argument library called TCLAP to parse the command line arguments.  I will take a look at this ASAP, but for the time being, maybe try -r=0.1?  Also, can you grab the version number of the arcwelder.exe binary you are using?  Run ```arcwelder.exe --help`` and  the version should look something like this:

```
Version: 1.1.0, Branch: HEAD, BuildDate: 2021-01-24T20:19:26Z
```
Thanks!

DarkAlchy, 2021-08-11:
The r value is irrelevant as it simply will not parse it.     Version: 1.1.0, Branch: HEAD, BuildDate:   2021-01-24T20:19:26Z

DarkAlchy, 2021-08-11:
I bet so.  I have seen this before many years ago where the command used was actually a command for the interpreter.  I wonder if TCLAP has something internal that uses -r?  It takes the long form so it is something with -r though I only used that so might be more than just that command too.

Thank you.

FormerLurker, 2021-08-11:
I appreciate the report, and will add a note about this in the help.  I will test the other parameters too.