FormerLurker/ArcWelderLib #74: Possible issue with print output after being processed by ArcWelder

CRCinAU, 2021-08-21:
Re: https://github.com/MarlinFirmware/Marlin/issues/22571

Could this issue in Marlin be why I see this on certain prints?

[image]

I'm not sure what else would cause the differences in the edge...

FormerLurker, 2021-08-21:
You are awesome 😎!  I obviously have lots on my plate, as you know, but will be be responding in detail with a few questions here soon.

CRCinAU, 2021-08-22:
Not awesome at all - these things are *waaaaay* above my head - but dumbing it down to me to understand, I think I understand the from a 10,000 mile high view.

As its likely to help, I'll attach:
STL: [Stand_1.stl.zip](https://github.com/FormerLurker/ArcWelderLib/files/7026735/Stand_1.stl.zip)
Sliced G-Code from PrusaSlicer 2.3.3: [Stand_1_0.2mm_PLA.zip](https://github.com/FormerLurker/ArcWelderLib/files/7026736/Stand_1_0.2mm_PLA.zip)
After processing with ArcWelder 1.1.0rc2 on Octoprint: 
[Stand_1_0.2mm_PLA.aw.zip](https://github.com/FormerLurker/ArcWelderLib/files/7026744/Stand_1_0.2mm_PLA.aw.zip)

Screenie of statistics:
[image]

Currently running:
FIRMWARE_NAME:Marlin bugfix-2.0.x (Aug 10 2021 01:53:14)

CRCinAU, 2021-08-22:
Ah - also, current config for that run might help:
[image]

FormerLurker, 2021-08-23:
Hey, can you post your marlin config (the adv) file?  This could be firmware settings related.

CRCinAU, 2021-08-24:
I don't have a copy of it for what I've built - however the base would have been from here:
https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration_adv.h#L2057

As such, those ARC settings would be used.

FormerLurker, 2021-08-24:
So you just enabled arc support and left the defaults in there?

CRCinAU, 2021-08-24:
Yep - I don't normally know what these do in detail, so I leave the defaults...

FormerLurker, 2021-08-24:
Ok, I did try all these files out, and I'm not sure what's going on.  The toolpaths look perfectly fine.  I can't tell from your pic if this is a toolpath problem, or an over/under extrusion issue.  Can you remember when you built this version?  Perhaps the algorithm has changed somewhat since you flashed it?

CRCinAU, 2021-08-24:
Yep - it was built on Aug 10 2021 01:53:14 from bugfix-2.0.x

FormerLurker, 2021-08-24:
Well darn.  That's pretty recent....  And you're using relative extrusion.  Have you printed the original (no g2/g3) gcode, and did it come out the same?

FormerLurker, 2021-08-24:
OH, and sorry for the flurry of questions.   These issues are often pretty perplexing.  Sometimes I figure out what's going on, lol!

CRCinAU, 2021-08-24:
To be honest, I haven't printed the non-AW'ed version - as my printing pipeline is basically PrusaSlicer -> Octoprint and then automatic AW processing.

Would it be helpful to try the same output with the same non-AW gcode posted here and see what happens?

FormerLurker, 2021-08-24:
> Would it be helpful to try the same output with the same non-AW gcode posted here and see what happens?

I hate to ask, but yes it would be very helpful.  If the same artifacts show up, then we can at least rule out arcwelder.  If they don't appear, it will be back to the drawing board.

CRCinAU, 2021-08-24:
It's printing now - so it'll take a few hours.... I think I might have reduced the feed rate on the original print I did, but as this seems to be sticking fine, I'm just letting it go at 100%...

FormerLurker, 2021-08-25:
Woot!  This is an ideal example of how a GitHub issue should work!  Thank you!  Am anxiously awaiting the results.

CRCinAU, 2021-08-25:
Well, not really - as after printing the non-AW processed file, it still has similar notches:

[image]

It does look like each of the 'notches' are level with par of the hex pattern in the rest of the print - as such, it probably isn't an AW issue....

I don't exactly know what it could be... Maybe linear advance isn't working the same on each corner? I dunno - but it doesn't seem to be AW and I've just been making noise ;)

ignisf, 2022-04-22:
Hi @CRCinAU, the bug you mentioned seems to be fixed in Marlin. Are you still experiencing the issue? (I am investigating an issue of my own which may or may not be this)

ignisf, 2022-04-22:
@FormerLurker, what is your preferred tool to review toolpaths with?

FormerLurker, 2022-04-22:
I use sumplify3d stand alone, ncviewer.com (some gcode modification may be required) for layer-by-layer analysis, and pretty gcode viewer for octoprint.

Also, this issue is common in marlin 1 forks, solved in latest marlin 2, if it is what I think it is.

CRCinAU, 2022-04-23:
Honestly, I forgot all about this and don't remember where I got to....

I figured that if it was happening without processing with AW, then its not related to AW...