FormerLurker/ArcWelderLib #80: ArcWelder console app for Windows removing perimeters for Klipper GCode.

slrAU, 2021-11-27:
Hi there,

Thanks for creating the ArcWelder tool. It is a really great concept and seems strange that none of the slicers are able to optimise gcode this well. From the very limited testing I've done it vastly decreases both file size and commands. The trouble is, each time I use it, all of the perimeters are removed from the models, which leaves me with some very rough-looking output. The next two images show the difference between the gcode output prior to post-processing with ArcWelder, and then after:

[image]

[image]

I've included the [gcode files](https://github.com/FormerLurker/ArcWelderLib/files/7611457/gcodeFiles.zip) gcode files used to create those images in SuperSlicer.

The ArcWelder console app version is 1.2.0-8-g71a516d, which I downloaded today direct from your compiled releases page.

The command that I used to generate the post-processed code was:
> arcwelder -l=VERBOSE ChristmasBellBase_0.24mm_PLA_ENDER3_6h8m.gcode weldedChristmasBase.gcode

This was executed on Windows 10 Pro 21H1, OS Build 19043.1348.

My slicer and printer are set up for the latest/current update of Klipper.

I don't think that it was anything I forgot to do at execution, and I would have assumed the default settings would apply arcs where it could and would leave lines where arcs need not be applied. I haven't yet tested far enough to see whether it's specific geometries or models that result in this issue. The model I've shown here seemed an ideal starting test as I both needed to print it, and it had lots of curved features to work with. 

I really hope you can figure out why this issue has appeared.  ArcWelder will be a huge time-saver without needing to overly tweak print settings, accessible even to novice users, and is a terrific idea that I would really love to see be completely successful. :)

Please let me know if you need additional information to assist you with your investigation.

FormerLurker, 2021-11-27:
Hi!  This is maybe the most courteous issue I've ever received :)

So, I took a look at the gcode.  The left image is the original gocde file, and the image on the right is the welded file from your zip:

[image]

Have you tried printing this file?  My guess would be (and this is EXTREMELY common) that the gcode visualizer within SuperSlicer does not support arc commands.  Since almost all of your perimeter has been converted to G2/G3 commands, it makes it look like they have been removed.

This is such a common problem, that I'm going to add some information about known working (and broken) gcode visualization tools.  The 'PrettyGcodeViewer' plugin for OctoPrint can correctly display G2/G3 (it bases the output on Marlin's arc interpolation algorithm, so it's quite accurate).  Also, Simplify3D (that's what I used to create the image above) can handle arcs.  OctoPrint's build in 2D visualizer also works.  I sometimes use NCViewer.com, but it has some issues.  It's confused by the embedded thumbnail images, so you have to remove those.  It's also confused by a couple gcodes that can cause rendering issues, but for the most part it works, especially if you only paste in a single layer.

I also ran the welded file through the 'ArcStraightener' algorithm.  I don't yet have a klipper simulator (I plan on adding one), but I know it's similar to Marlin 1's algorithm.  Be sure to turn the mm_per_arc_segment setting down to 0.5, or 0.1 (lower is better, to a point, as long as your hardware can handle it).  Here's the result:

[image]

So, at least Marlin will correctly translate the arc movements.  I'd recommend giving the welded file a try and see how it goes.  It looks like all is well.

FormerLurker, 2021-11-27:
FYI, I added information regarding working visualizers [here](https://github.com/FormerLurker/ArcWelderLib/#gcode-visualizers-that-do-not-support-g2g3-commands).

slrAU, 2021-11-28:
Hi again and thanks for such a fast and thoughtful response.

I really should have mentioned that I did indeed try to print the model. I let the printer run through a couple of layers first before concluding that the output was going to be exactly the same as the visualizer.  

Interestingly I ran across [this issue](https://github.com/supermerill/SuperSlicer/issues/1456) on the SuperSlicer Github where you got a mention, and apparently G2/G3 are meant to have been supported.  That's not such a big deal though as I can always use a different visualizer if I really need it.

Given you've had no trouble with visualizers that properly support G2/G3, I'm starting to think that I probably have a firmware configuration issue as well. I thought that the arc commands had been enabled when I last built Klipper for my printer, but clearly I must have missed something and should probably try and adjust the firmware again.

In any case, thanks again for your insight. I'm really looking forward to getting ArcWelder properly integrated into my workflow. ;)

FormerLurker, 2021-11-28:
Not a problem.  I will look into klipper config and see if maybe I can assist getting arcs working for you

slrAU, 2021-11-28:
I found it! 

It was a combination of my poor eyesight and a subtle typo in the end. I'd transposed a couple of letters in the Klipper config file  which effectively left the arcs feature "disabled".  I'm running a test print now and it seems to be obeying the commands well, so I'd say the issue is resolved. Sad to say Human Error the cause! I keep telling my kids to take a step back and have another look at a problem, so it's Karma sending my own words back to haunt me! LOL

Thanks for being so understanding.

closed 2021-11-28 by slrAU