FormerLurker/ArcWelderLib #81: [FR] Add switch to place comment at end of G2/G3 to show number of G1/G0 moves 'absoe

CmdrCody51, 2021-11-28:


FormerLurker, 2021-11-28:
Interesting.  Can I ask why you want to see this?

CmdrCody51, 2021-11-28:
Oh great - it didn't (or I didn't) save my comment......
I would like a command line switch to add a comment to the Arc commands counting the number of linear moves replaced by the arc. I use ArcWeldLib as a stand alone script from SuperSlicer so I can't really have any stats. I think it would be easier to fine tune the paramerters if I knew how well ArcWeldLib is running. OctoPrint strips comments anyway so it doesn't affect the print. I have seen on a 'pure circle' outputs of three separate arcs and have wondered what made the 'break'.

CmdrCody51, 2022-05-03:
Well now,  darn near forgot about this...
Actually I was looking for a per line break down kinda like:
G1 X100 Y100
G2 X110 Y110 I0 J-10 ; count 15

That way no extrusions would have no E.
I know ArcWeld works but some times an arc is not made when I think it should be.
Especially on brims.
Over all stats are good but I wanted something more granular.