FormerLurker/ArcWelderLib #85: Windows binaries not statically built

fieldOfView, 2022-01-13:
The Windows builds of version 1.1.1 and 1.2.0 are statically built (ie: it depends on the visual c++ runtime). This causes the binaries to fail with a returncode 3221225781 on systems where the right version of the visual c++ runtime is not installed.

The 1.1.0 build was statically built, so it runs regardless of whether the visual c++ runtime is installed.

fieldOfView, 2022-01-13:
No problem, take your time. I'm just reporting it so you can fix it for 2.0.

FormerLurker, 2022-04-26:
So, I'm finally able to continue working on things.  This is my next priority.  Just wanted to let you know.

ETE-Design, 2022-05-23:
@FormerLurker How is work going for now?

FormerLurker, 2022-06-07:
Well, poorly.  I spent the several weeks out of town helping my mom and dad take care of my grandmother, who passed away during that time + funeral.  I'm totally OK, but this left me unable to finish this up.  I promise I've been working on this though, and will get it done.  I have some other projects I'd like to start as well, but need to get ArcWelder (priority 1 for arachne support) and Octolapse (priority 2) squared away before that.

fieldOfView, 2022-06-08:
> I'm totally OK,

Glad to hear that. This may sound presumptuous, but take the time you need, there are more important things than code.