FormerLurker/ArcWelderLib #88: Imperfect G2/G3 generated for full circles

fabien, 2022-06-07:
As outlined [here](https://github.com/FormerLurker/ArcWelderLib/issues/70#issuecomment-1148269124) I'm using ArcWelder to prepare files for CNC cutting using a tangential knife. Apart from the former issue, the generated G2/G3 code looks pretty good. 

However, perfect circles always have one larger arc (say, 90% - 95% of the perimeter) and end in a small arc or line segment. I don't think this will cause issues cutting, but I presume it should be possible to turn a perfect circle into a single G2/G3 instruction. I have tried different settings, but the circle output remains segmented in two arcs.

See: https://gist.github.com/fabien/a4d82a31fd9c1a58586d5cd45c170d42

[image]
[image]

FormerLurker, 2022-06-07:
@fabien, 

I may have to add an option for this, as full circles mess up 3d prints, resulting in an unacceptable overlap of filament.  There is actually code to prevent full circles in arcwelder currently.

I'll think about this, and will mark it as an enhancement (including changing the title).

Thanks!

fabien, 2022-06-07:
@FormerLurker I didn't know it was a feature ;) (to prevent full circles)

Looking forward to such an enhancement (an option like `--allow-circles`?) Thanks for this extremely useful lib/tool!

PS will you be releasing 2.0 sometime in the near future? Hoping [this](https://github.com/FormerLurker/ArcWelderLib/issues/85) will be resolved as well, as ultimately, I need to run it on Windows too.

FormerLurker, 2022-06-07:
You can use it on windows (that's what I use to develop the tool).   You just need to [install the c++ runtimes first](https://docs.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist?view=msvc-170).

FormerLurker, 2022-06-07:
Crap...  I tell you what, give me thru the weekend and I'll try to push a build up.  I'm in the middle of some modifications, and can't push the changes till those are finished.  Will try to get it even sooner if I can.

ETE-Design, 2022-09-13:
@FormerLurker V2 seems to still be Expired :-( Can you upload it again?