FormerLurker/ArcWelderLib #90: arc welder implementing feedrate changes at end of move

SrgWhoopsie, 2023-01-01:
not sure if this is to do with the arcwelder itself or if its the cura plug in or if its cura 5.2.1, but when viewing the sliced gcode with zupfe i noticed there were extrusion moves at 150mm/s as it hadnt included the change from travel to extrusion, when viewing the code in vscode i noted in various points of my gcode, through out multiple sliced items feedrate changes were being added to the end of the movement instead of the start of the movement .
[image]
[image]

Kathode-Ray, 2025-09-16:
I'm having the exact same problem, i'm using Cura 5.10.1 and Arc Welder plugin 3.6.0.

The F<rate> parameter is inserted at the end of the G2/G3 code line, so it is not processed. The next lines are printed with the incorrect (previous) speed. This is a screenshot of a simple model with printspeed set at 15mm/s for the first layer.

[image]

Kathode-Ray, 2025-09-16:
After a bit of digging around I decided to try this with Orca Slicer to see if the problem is also present there.

With Orca Slicer, the formatting of the gcode lines is slightly different. First, there's a G1 command that sets the speed to 20mm/s and in the next line the arc is drawn:

[image]

When using Cura with the Arc Welder plugin, the speed / rate is included as a parameter in the G2/G3 code line:

[image]

So, i'm wondering if this caused by tha Arc Welder plugin, or is it the ZupFe gcode viewer that interprets the gcode not as it should?
Can anyone shed some light on this?