FormerLurker/ArcWelderLib #92: Issue when material diameter is set to large values in Cura

Bikingcrazy52, 2023-04-20:
Hi,

I use Cura to print to make toolpaths for a custom printer that uses material approximately 22 mm in diameter, thus extrusion speeds are much slower than with 2.85 mm filament.

The issue is that ArcWelder kinda falls apart when the material diameter is set to these values.

For example, if i create a simple STL that is just a 10 mm diameter disk and slice it with ArcWelder enabled and a standard material diameter of 2.85 selected, it will create one nice 10 mm arc for the OD.

If all I do is change the material diameter in machine settings-->Extruder 1-->Compatible material diameter: 22 mm ( and I also made a custom material that is 22 mm in diameter) ArcWelder creates like 10 arcs and segments to make what should be one.

Any ideas?

FormerLurker, 2023-04-20:
Please provide your original gcode (not processed by arcwelder) and the
processed file.  I recommend using gist.github.com for this.  Then drop in
a link to the gist.  Thanks!

Bikingcrazy52, 2023-04-21:
Here ya go. Also included the same part processed with ArcWelder where the only thing I changed is the material diameter back to the standard 2.85 mm, which looks like it processed it correctly. Thanks for the help!

https://gist.github.com/Bikingcrazy52/783a0b9b686eebc7becd29ecac2b2009

Bikingcrazy52, 2023-04-24:
I think I figured it out. Because of the super small extrusion commands the extrusion rate variance parameter in ArcWelder settings had to be set to 10% or higher, or 0 to disable, to get it to process the arcs. Kinda makes sense.

Thanks!